FreeCAD/FreeCAD #13577: TechDraw: Editor for general page layout (ISO 5457) and title block template (ISO 7200)

maxwxyz, 2024-04-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently there are .svg for each language, page size, page orientation and special usecase.
It would be great if there could be a way to edit the page template in TechDraw / FreeCAD directly which could be adaptive to page sizes.
1. Option to define a title block (ISO 7200) and add all items you need, define it's justification, where new cells for a history are added and so on. So it would be independent of the page size and language. Ability to add fields like projection method or custom fields and link them to document properties.
2. Option for a template editor for the general page layout (ISO 5457), define grid, orientation, position of the template for the title block (e.g. always bottom right, to be independent of the page size and orientation)

This could also solve: 
- https://github.com/FreeCAD/FreeCAD/issues/13475
- https://github.com/FreeCAD/FreeCAD/issues/13539

.svg files could stay supported as well.

From: https://github.com/FreeCAD/FreeCAD/issues/11671

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.34
  * fasteners 0.5.17
  * freecad.gears 1.2.0
  * OpenTheme 2024.4.20
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ab0v3g4me, 2024-04-21:
I just want to put this out there as well as it might be related: 

[Modular table display](https://forum.freecad.org/viewtopic.php?t=86899&sid=0d93fcf4c9eed5541645a53b0e47358c)

tobiasfalk, 2024-04-21:
I don't want to be to shamelesss, but I made a TempGen and I am currently working on a complete overwhole of it, in branch Vers.-2.
[](https://github.com/tobiasfalk/KiCAD-FreeCAD-TechDraw-Template-Generator)

maxwxyz, 2024-04-21:
@tobiasfalk interesting, never heard of it before! Any chance this would be available as addon or merged directly into FreeCAD?

tobiasfalk, 2024-04-21:
I already hat the idea to make a Integration in to FreeCAD but I also want to  Keeper it as a separate Tool. 

At the Moment the main aim is to get V2 to the base functinalety with a simple frame and Titleblock.

After that as ISO 5456 and some ISO 7200 Titleblocks.

At this point I would have no Problem if the FreeCAD Team wants to take over the Projekt and integrate it in to The FreeCAD domain.

One thing where I have no Idea how to do it, is how it would be posible that the core is written in C++ and the general design/template defining code is in Python.

maxwxyz, 2024-04-21:
@tobiasfalk cool!
TechDraw is also written in Python, same as Draft WB.

pure-machinery, 2024-04-21:
@tobiasfalk Very nice! I am also working on an API for generating svg templates. 

Here's a function that draws my base table:

<details>
<summary>Example code</summary>

```odin
DrawBaseTable :: proc(width, height, left_offset, offset: f32, page_size: string) {
	st := svg_style {
		flags = { .FILL, .STROKE },
		stroke_width = 0.5,
		stroke_color = { 0.0, 0.0, 0.0, 1.0 },
	};	

	gr := svg_group { id = "base_table"};
	BeginElement(.GROUP, &gr, &st);

	TABLE_HEIGHT :: 60; 
	TABLE_WIDTH :: 185; // Fits A4 with left offset of 20 mm and right offset of 5 mm.

	DESC_FONT_SIZE :: 3;
	EDIT_FONT_SIZE :: 5;
	TOTAL_ROW_HEIGHT :: (DESC_FONT_SIZE + EDIT_FONT_SIZE) * 1.5;

	r := svg_rect { 
		width - offset - TABLE_WIDTH, 
		height - offset - TABLE_HEIGHT,
		TABLE_WIDTH,
		TABLE_HEIGHT,
	};

	a := RectCut(&r, .TOP, TOTAL_ROW_HEIGHT);
	b := RectCut(&r, .TOP, TOTAL_ROW_HEIGHT);
	c := RectCut(&r, .TOP, TOTAL_ROW_HEIGHT);
	d := RectCut(&a, .RIGHT, TABLE_WIDTH * 0.5);
	
	e := RectCut(&b, .RIGHT, TABLE_WIDTH * 0.5);
	f := RectCut(&e, .RIGHT, TABLE_WIDTH * 0.5 * 0.5);

	g := RectCut(&c, .LEFT, TABLE_WIDTH * 0.25);
	h := RectCut(&c, .LEFT, TABLE_WIDTH * 0.25);
	i := RectCut(&c, .LEFT, TABLE_WIDTH * 0.25);

	BeginElement(.RECT, &a, nil);
	EndElement(.RECT);
	BeginElement(.RECT, &b, nil);
	EndElement(.RECT);
	BeginElement(.RECT, &c, nil);
	EndElement(.RECT);
	BeginElement(.RECT, &d, nil);
	EndElement(.RECT);
	BeginElement(.RECT, &e, nil);
	EndElement(.RECT);
	BeginElement(.RECT, &f, nil);
	EndElement(.RECT);
	BeginElement(.RECT, &g, nil);
	EndElement(.RECT);
	BeginElement(.RECT, &h, nil);
	EndElement(.RECT);
	BeginElement(.RECT, &i, nil);
	EndElement(.RECT);

	EndElement(.GROUP);

	temp_st := svg_style {
		flags = { .FILL, .TEXT },
		fill_color = { 0.0, 0.0, 0.0, 1.0 },

		text_color = { 0.0, 0.0, 0.0, 1.0 },
		text_font = "FreeSans",
		text_size = DESC_FONT_SIZE,
	};	

	gr.id = "placeholder_text";
	BeginElement(.GROUP, &gr, &temp_st);

	PlaceText(&a, "Description:", DESC_FONT_SIZE);
	PlaceText(&b, "Designation:", DESC_FONT_SIZE);
	PlaceText(&e, "Created by:", DESC_FONT_SIZE);
	PlaceText(&f, "Date created:", DESC_FONT_SIZE);
	PlaceText(&g, "Mass [kg]:", DESC_FONT_SIZE);
	PlaceText(&h, "Scale:", DESC_FONT_SIZE);
	PlaceText(&i, "Sheet size:", DESC_FONT_SIZE);
	PlaceText(&c, "Sheet No. :", DESC_FONT_SIZE);

	EndElement(.GROUP);

	temp_st.text_size = EDIT_FONT_SIZE;
	gr.id = "edit_text";
	BeginElement(.GROUP, &gr, &temp_st);

	toffset : f32 = DESC_FONT_SIZE + EDIT_FONT_SIZE + 1.0;
	PlaceText(&a, "title", toffset, { .EDIT, .AUTOFILL });
	PlaceText(&b, "pdesg", toffset, { .EDIT });
	PlaceText(&e, "author", toffset, { .EDIT, .AUTOFILL });
	PlaceText(&f, "date", toffset, { .EDIT, .AUTOFILL });
	PlaceText(&g, "pmass", toffset, { .EDIT });
	PlaceText(&h, "scale", toffset, { .EDIT, .AUTOFILL });
	PlaceText(&i, page_size, toffset);
	PlaceText(&c, "sheet", toffset, { .EDIT, .AUTOFILL });

	EndElement(.GROUP);

	gr.id = "logo";
	BeginElement(.GROUP, &gr, nil);

	IMAGE_WIDTH :: 37;
	IMAGE_HEIGHT :: TOTAL_ROW_HEIGHT; 

	x_offset := d.x + d.width * 0.5 - IMAGE_WIDTH * 0.5 ;
	y_offset := d.y + 1.0;

	fmt.println(x_offset, y_offset);

	img := svg_image {
		src = "/home/abovegame/Documents/DrawingLayouts/logos/logo_pm.svg",
		x = x_offset,
		y = y_offset,
		width = 39.480,
		height = 10.500,
		keep_aspect_ratio = true,
	}

	BeginElement(.IMAGE, &img);

	EndElement(.GROUP);
}
```
</details> 

It could definitely use some more polishing. I might share it on the forums once I'm satisfied with it.

WandererFan, 2024-04-21:
> TechDraw is also written in Python, same as Draft WB.

Most of TechDraw is in c++.  There are only a few tools done in Python (in Mod/TechDraw/TechDrawTools).

Integrated a Python tool with a c++ base shouldn't  be a problem.

tobiasfalk, 2024-04-24:
@maxwxyz the base functionality of a Simple Frame and a very simple TitleBlock(that has no Text LOL) is done. 
I am now starting on a more advanced Frame and TitleBlock.
As mentioned if there is interest in integration this in to FreeCAD TD, I would say the best way would be as a git subproject, and then it is compiled into FC, this way it can still be as its own Tool.
When this is the case, I would have no problem with putting this under the FreeCAD domain and name, maybe with a name change to "Universal Template Generator"(UTG) or so.

tobiasfalk, 2024-04-24:
I think that an Integration in to FreeCAD should not be too difficult and is most likely the only option since my Tool is not written in Python and the addon manager is pure Python based.

tobiasfalk, 2024-05-03:
I have finished the basics that I mentioned before, that includes a ISO 5457 frame and two ISO 7200 comform Titleblock.
As to integration in to FreeCAD, in my opinion the the best was would be to not use the FreeCAD SVG output but rather the fact that I included the ability to directly draw on to a QPaint Object. This could also be done without opening the TemplateGen Window bur rather the selection of the Frame, Titleblock, .... Style through a Template Object, maybe with the ability to "export" this Object as a KiCAD Template or the other Formats that are available.

maxwxyz, 2024-05-03:
sounds great!

maxwxyz, 2024-07-06:
Other idea: would it be interesting to make the TD templates available via the addon manager? This way they could also be updated without a new release of FreeCAD.

furgo16, 2024-07-06:
That would also be useful for translators