FreeCAD/FreeCAD #13581: Draft Mirrored body fails to export to Step file.

czinehuba, 2024-04-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying to export one of my projects and the mirrored PD body doesn't get exported into the step. None of the mirrored bodies are showing up in the assembly. Not using any assembly workbench, just placement for bodies and subassemblies.

Steps for reproducing the issue:
- Add Part to a new file
- Add a body to the part. 
- Create a primitive (or any other operation in PD.)
- Then use draft WB to mirror the body. 
- Drag the mirror onto the part, so the mirror consumes the body.
- Then export the whole Assy into a step file. 
Only one of the bodies will be in the step file.

I've tried it with FC 21.2, RT and dev versions. All do the same thing.

[mirror bug.zip](https://github.com/FreeCAD/FreeCAD/files/15053330/mirror.bug.zip)

### Full version info

```shell
OS: Ubuntu Core 22 (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36913 (Git) Snap 994
Build type: Release
Branch: main
Hash: 700868f8346478f88e955672e5e1bad47ae061da
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * fcgear 1.0.0
  * Curves 0.6.34
  * fc_nodes 0.1.36
  * sheetmetal 0.4.11
  * fasteners 0.5.17
  * lattice2 1.0.0
  * freecad_metal_workbench 0.0.1
  * ExtremeProLight 1.6.0
  * Nodes 0.1.36
  * CfdOF.backup1713725901.4016135 1.21.7 (Disabled)
  * freecad.gears 1.2.0
  * OpenTheme 2024.4.20
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

vnen, 2024-08-03:
I have this issue with 0.21.2 but I noticed that the mirrored part is the one present in the STEP file, while the original one is missing.

eblanshey, 2025-04-23:
I have the same problem as @vnen in the latest v1.1dev build. The mirrored part is exported but not the original. Probably the export type doesn't matter, but I'm using .gltf.

eblanshey, 2025-04-23:
I just noticed that the same issue occurs if you select the part and make a subshape binder. The binder has the same missing parts (originals of the mirrors).

eblanshey, 2025-06-26:
Confirmed still occurring on the latest dev version. The original of the mirror is not being exported for a part container with many parts and mirrored parts.

```
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42325 (Git) AppImage
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/140.99/1
Installed mods: 
  * WebTools 1.0.0
  * Curves 0.6.58
  * Plot 2024.11.26
  * AirPlaneDesign 0.4.1
  * Help 1.0.3
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
```

czinehuba, 2025-08-27:
Yes. The problem still persists on the latest dev version. 

OS: Ubuntu Core 22 (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.43088 (Git) Snap 1675
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/138.249/1
Installed mods: 
  * freecad_metal_workbench 0.0.1
  * fasteners 0.5.38
  * fcgear 1.0.0
  * OpenTheme 2025.5.20
  * Curves 0.6.62
  * freecad.gears 1.3.0
  * CfdOF.backup1713725901.4016135 1.21.7 (Disabled)
  * lattice2 1.0.0
  * sheetmetal 0.7.22

It exports correctly when clicking the legacy exporter.