FreeCAD/FreeCAD #13583: Addon Manager - Useless Checking Connection window at first launch

marcuspollio, 2024-04-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi there,

When launching the Addon Manager (Tools > Addon Manager) for the first time (e.g. after a fresh config), a connection check and download of external addons metadata is performed. This process is displayed in the Addon Manager window that just popped up.
However, a third window called "Checking connection" is opened in the background. This third window cannot be brought the foreground, at least on my system, and stays open a long as the Addon Manager window is open too.
When trying to reproduce later on other FreeCAD instances, this third window does not open anymore.

### Full version info

```shell
OS: GNU/Linux distro based on Ubuntu 22.04 LTS jammy (DE:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36915 (Git)
Build type: Release
Branch: main
Hash: 1f9be3f9ed44f7a3f4a453ef4761eea2d96a3dc0
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
```

### Subproject(s) affected?

Addon Manager

### Anything else?

Similar post in the Forum :[Network Manager Error](https://forum.freecad.org/viewtopic.php?p=738745#p738745)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-22:
@chennes FYI

marcuspollio, 2024-06-04:
Closing as it's not reproducible anymore with :

```
OS: GNU/Linux Ubuntu 22.04 LTS (DE:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37652 (Git)
Build type: Release
Branch: main
Hash: 36a7fd9be42f2b546562fae832446e8420f9db0c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.6.3
```

closed 2024-06-04 by marcuspollio