FreeCAD/FreeCAD #13597: DXF importer no longer imports layers other than the "0" layer

furgo16, 2024-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The DXF importer in the development branch seems to no longer create layers correctly. I've seen it working at some point, but now only layer 0 seems to be imported correctly. The stable branch has some sort of layer support, and the results are also shown here for comparison. 

The starting point is a [sample file](https://github.com/jscad/sample-files/blob/master/dxf/dxf-parser/layers.dxf) that contains 3 layers, with a simple object (rectangle or circle) in each layer. The drawing looks like in the following screenshot, as loaded by QCAD:

[image]

## DXF import with FreeCAD 0.22

### As single elements

The following settings were used:
[image]

The result:

- [ok] Single elements are imported correctly, including colors. They use specific names (line, circle) according to their shape.
- [**fail**] Only layer 0 is imported as a layer. The rest of elements outside of 0 appear layerless

[image]

[dxf_layers_single_dev.zip](https://github.com/FreeCAD/FreeCAD/files/15074375/dxf_layers_single_dev.zip)

### As layer blocks

The following settings were used:
[image]

The result:

- [ok] Grouped elements are imported correctly as compounds, including colors. They a generic name (CompoundNNN).
- [**fail**] Only layer 0 is imported as a layer. The rest of elements outside of 0 appear layerless. However, each block is correctly created as a container for the elements that would belong to a specific layer. This was a similar behavior to FreeCAD 0.21, but 0.21 had the ability to name the block/compound as the layer it had been constructed from (see 0.21 example below).

[image]

[dxf_layers_layerblocks_dev.zip](https://github.com/FreeCAD/FreeCAD/files/15074435/dxf_layers_layerblocks_dev.zip)

## DXF import with FreeCAD 0.21

The following comparison was made using FreeCAD's stable version:

```shell
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) Snap 908
Build type: Release
Branch: tag: 0.21.2
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Curves 0.6.23
  * Assembly4 0.50.7
```

### As single elements

The following settings were used:
[image]

- [ok] Single elements are imported correctly, without colors. They use generic names (ShapeNNN).
- [**fail**] No layer grouping at all

[image]

[dxf_layers_single_stable.zip](https://github.com/FreeCAD/FreeCAD/files/15074516/dxf_layers_single_stable.zip)

### As layer blocks

The following settings were used:
[image]

- [ok]  Grouped elements are imported correctly as compounds, including colors. Compounds are named after the layer it originally contained all their elements.
- [**fail**] No layer grouping, but each block is correctly created as a container for the elements that would belong to a specific layer. However, the naming of the compounds using the original layer name makes it possible to identify and use the original layers

[image]

[dxf_layers_layerblocks_stable.zip](https://github.com/FreeCAD/FreeCAD/files/15074606/dxf_layers_layerblocks_stable.zip)

## Conclusion

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36961 (Git) Snap 996
Build type: Release
Branch: main
Hash: c4c453312d148fd67bd893366c42b559e28a85e6
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-04-24:
Also related to imported DXF layers: https://github.com/FreeCAD/FreeCAD/issues/13621

furgo16, 2024-04-27:
I can no longer reproduce this issue. Today I could import layers successfully with the C++ importer. I wonder if the fix for https://github.com/FreeCAD/FreeCAD/issues/13636 had something to do with it. I'll keep monitoring, as I'll be doing more imports in the next few days, and if I still cannot reproduce it, I'll close the issue.

maxwxyz, 2024-06-24:
I'll close it then.

kpemartin, 2024-09-08:
I've started looking at this as a simpler example of #16068 and I'm finding that opening the same file several times (without restarting FreeCAD) and the same options results in different results. Most of the time I get the outer box in layer 0, and the inner box and circle at top level. (with Use Layers, no combining). Just once I got the outer box in layer 0, the inner box in layer dashedred, and the circle still at top level.

kpemartin, 2024-09-08:
The broken layers, at least when not joining the objects, seem to be occurring because a C++ call to `PyImport_ImportModule("Draft")` sometimes works and sometimes doesn't.

This is the same sort of odd behaviour I saw while verifying my fix for #16131 (where `Document.addObject("Part::FeaturePython", "xxx")` was eventually trying to load the `Part` module and failing for no apparent reason). I still don't know if it is my build environment or something else.

furgo16, 2024-09-10:
I can reproduce the issue again with recent builds. Given this and @kpemartin's feedback, and that there wasn't a previous PR that fixed it, @maxwxyz could you please reopen the issue? Thank you.

furgo16, 2024-09-10:
@kpemartin do you have any draft branch we could help testing? Thanks a lot for your work.

kpemartin, 2024-09-11:
I'm putting together a PR to partially address #16068 which also includes a fix for remediation for the problem in this Issue that some of the layers are missing and the drawing objects in them instead appear as top-level objects in the FreeCAD document.

I, too, think this should be re-opened so that my PR can definitively mark it fixed.

The fix is to realize that the `CDxfRead` class already has a reference to the `Draft` module so `ImpExpDxfRead::MakeLayer` doesn't have to fetch it again. Also the code in `CDxfRead` that obtains this reference issues an error if it fails, to remove the element of mystery when things start acting strange.

When I use this branch to build I can reliably get the other problem I mentioned above, one of the lines missing from the drawing (specifically, the bottom of the red square), or one of the compounds missing (I don't recall which one) if "group layers into blocks" is selected. Essentially the first call to `doc->addObject("Part::FeaturePython", "xxx")` throws a C++ exception (a PyException) because of a module import error. I'm considering making a fresh issue out of this because of the complexity of the analysis which would otherwise just clutter up this Issue and #16068.

kpemartin, 2024-09-11:
I suspect the module import problem may be an environment/configuration issue so it may only show up in certain situations, which could lead to difficulty in reproducing it.

The code branch I'm using is only a few lines changed (the fixes mentioned above) from `master`.

I'm on Windows 10, developing using MS Visual Studio 2022 (currently 17.11.2), CMake 3.30.3, libpack 1.0.0-v3.0.0-Release.

CMake configuration has been changed to point `FREECAD_LIBPACK_DIR` to my libpack location, and to turn on `FREECAD_COPY_DEPEND_DIRS_TO_BUILD`, `FREECAD_COPY_LIBPACK_BIN_TO_BUILD`, and `FREECAD_COPY_PLUGINS_BIN_TO_BUILD`

I'm using the 'Release' build because the libpack won't work with the 'Debug' build, and 'RelWithDebInfo' places its build output in a different folder than 'Release' and 'Debug', a place that does not receive the copies from FREECAD_COPY...TO_BUILD, making it somewhat useless (and I can't figure out how to change this). This in turn makes debugging difficult because of code and variable optimization. Also VS doesn't support Python 3.12.

One oddity in my setup: Because of limited space on my system drive (an SSD) my user Documents folder is on a different hard drive, and C:\users\kpmartin\Documents is a windows junction point (hard link) to D:\users\kpmartin\Document and I have a minor thought that this C:/D: duality with otherwise identical paths is confusing things. As far as I can manage, though, all my build/run environment uses the D: paths exclusively.

kpemartin, 2024-09-12:
Found it, nothing to do with my setup. details and fix to follow.

commit 288b6e69 (2024-09-23): DXF: Fix Import related behavior reported in issues #13597 and #16068 (#16511)

* Fix double-import on exception in ImportGUI.readDXF
The python code has a try/catch block intended to detect if the ImportGUI module is present and if so use its readDXF method, otherwise use Import.readDXF.
The block was incorrectly structured so that Import.readDXF would also be called if ImportGUI.readDXF raised an exception, causing the DXF file contents to be loaded twice into the drawing (at least, up to the point where the exception occurred)

* Make ImpExpDxfRead::MakeLayer use centralized accessor for Dr [... cut]
closed 2024-09-23 by yorikvanhavre