FreeCAD/FreeCAD #13598: DXF importer ignores the "Create as" import settings and always creates simple Part shapes

furgo16, 2024-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The DXF importer preferences present the following "Create" options to the user, as shown in the screen capture below:

- simple Part shapes
- Draft objects
- Sketches

[image]

However, regardless of the complexity of the DXF and the option chosen, the resulting FreeCAD objects created after the import are always simple Part shapes. I have tested this with multiple files and settings, and while I might have not been exhaustive, I never managed to obtain Draft objects or Sketches.

This can be tested with a [sample DXF file](https://github.com/jscad/sample-files/blob/master/dxf/dxf-parser/layers.dxf) that contains 3 layers, with a simple object (rectangle or circle) in each layer. The drawing looks like in the following screenshot, as loaded by QCAD:

[image]

Steps to reproduce:

1. Go to Preferences > Import-Export > DXF
2. Under Create, select either "Draft objects" or "Sketches"
3. Import the [sample DXF file](https://github.com/jscad/sample-files/blob/master/dxf/dxf-parser/layers.dxf)
4. Expected: the imported components appear as "Draft objects" or "Sketches" (depending on the selection on step 2) in FreeCAD
5. Actual: the imported components are always "simple Part shapes"

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36961 (Git) Snap 996
Build type: Release
Branch: main
Hash: c4c453312d148fd67bd893366c42b559e28a85e6
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-04-23:
You have perhaps not noticed the text: "Note: Not all the options below are used by the new importer yet". In other words you need to switch to the legacy importer to use these options.

furgo16, 2024-04-23:
Oh, I had read that note. I mistakenly assumed it referred to the options marked as `(legacy importer only)`.

Roy-043, 2024-04-25:
Oops, you have got a point there. I'll add that text then. With my PR I'll close this issue. It is a duplicate of (or is covered by) #13599.

PR #13638 (2024-05-09): Draft: Add "legacy importer only" to DXF option
Fixes #13598.
commit 2f9b97ce (2024-05-09): Draft: Add "legacy importer only" to DXF option (#13638)

* Draft: Clarify "legacy only" DXF options in the preferences

Fixes #13598.
closed 2024-05-09 by Roy-043