FreeCAD/FreeCAD #13599: There should be a single DXF importer/exporter

FEA-eng, 2024-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

DXF import/export is a hot topic. There have been many issues with it and it seems that there are still some problems left. Apparently, the legacy importer/exporter should be used in most cases. But ideally, there should be only a single importer/exporter (with all the benefits and features of the current two implementations) or at least the better/recommended one should be the default to avoid the confusion. Are there any plans for that? I guess that it won't happen before the next release but it's something that should be resolved as soon as possible, in my opinion.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git)
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-04-23:
Ideally the newer, built-in C++ dxf importer should be used but it still does not support all the features of the old one... Not sure how fast this can be done, it basically needs volunteers to work on it :)

FEA-eng, 2024-04-23:
> Ideally the newer, built-in C++ dxf importer should be used but it still does not support all the features of the old one... Not sure how fast this can be done, it basically needs volunteers to work on it :)

Thanks for the reply. Sure, there's no hurry but I just wanted to document this for the future and ask if it's already planned.

Maybe a potential GSoC candidate?

furgo16, 2024-04-23:
> it still does not support all the features of the old one

Is there a list of missing features, or a side by side comparison of the two existing importers/exporters? If there isn't, how how could a community member best start compiling such a list?

Syres916, 2024-04-23:
Purely from a personal point of view, my defacto goto solution to get a dxf file into FreeCAD is https://pypi.org/project/ezdxf/ it's extremely well maintained in my experience, handles layers well and I just run a macro to convert all the objects to sketches.

furgo16, 2024-04-23:
> I just run a macro to convert all the objects to sketches.

Out of interest, which macro do you use to talk to ezdxf? I've noticed a similar approach at https://forum.freecad.org/viewtopic.php?t=81962, but I've not tried it.

Syres916, 2024-04-23:
I use https://github.com/FreeCAD/FreeCAD-macros/blob/master/Conversion/DxfToSketchLayers.FCMacro but I've not tested it on recent 0.22dev so it might need modifying. Ignore the installation part of the code, just install _ezdxf_ using _pip_ before running the macro.

yorikvanhavre, 2024-04-24:
It depends a lot on your given workflow too... DXF files are used for sometimes very different purposes. In the construction field, DXF files with thousands of small objects are not a rarity, while for people working with CNC a DXF file is often one single, precise profile shape. So for some users it's important that files import as a clean sketch, for others that would be counter-productive

kpemartin, 2024-09-15:
Some notes regarding that table:
Dimensions and text are perforce Draft objects, there are no "shape" counterparts.
Some groundwork for both creating Draft objects and joining geometry were done as part of #7551.
Some things like "Render Polylines with width" are really only partly done, IIRC. Polylines with fixed width import with the legacy importer but ones with variable width don't get the varying width.

My personal goal with the C++ importer is to be able to migrate old drawings to FreeCAD and have them be as usable and editable as in the original drawing app.

kpemartin, 2025-01-17:
I'm not sure either way of what the consequences would be of including this in GCoC 2025. For that matter, I'm not sure if I would be considered a mentor (at least for the C++ DXF importer) or a "mentee"!

maxwxyz, 2025-01-17:
@furgo16 done. Would this unified importer also include DWG conversion?

furgo16, 2025-01-17:
> I'm not sure either way of what the consequences would be of including this in GCoC 2025. For that matter, I'm not sure if I would be considered a mentor (at least for the C++ DXF importer) or a "mentee"!

I think either way your contribution, as far as I've seen, would be excellent! :-) As for the consequences, this is simply a proposal I'm putting forward as a contributor and regular user of the importers (I always have to swap between them and ezdxf or combine them to get the imports right). First the FreeCAD maintainers need to decide if it's worth considering, and then, if FreeCAD is accepted as an organization for GSoC this year, someone would have to pick the project. Then the goal would be for a mentee together with a mentor to complete the agree scope of the project by the time and rules set by the program: https://summerofcode.withgoogle.com

Having managed GSoC applications for other organizations, the level of success in getting a project completed is affected by a number of factors. In my opinion, those to highlight would be: how well the project is defined, whether it's attractive to potential contributors, and the level of committment/availability of both the mentor and the mentee.

> Would this unified importer also include DWG conversion?

@maxwxyz essentially, I'd propose to merge the two existing DXF importers (legacy and new) into one for a better user experience. The definition of merging can be better fleshed out if there is interest, but the consensus seems to be for the new (C++) importer to be the one to prevail, and either gain feature parity with the old or evolve to have the right set of features for an optimal import workflow.

DWG import is a project on its own and I'd consider it to be out of scope. That would be another project on its own, which is not straightforward. For instance, as as far as I understand, the only open source DWG converter (libreDWG) cannot be distributed with FreeCAD. I'm not an expert on this though. As a workaround, I always convert .dwg files with QCAD to .dxf and then I import those to FreeCAD.

kpemartin, 2025-01-27:
A couple of other things that the legacy importer has some support for but not the C++ importer:
- `block_record` table in the `TABLES` section. It isn't clear from the (2018) DXF file spec what these do, but I assume they were included in the legacy importer because someone wanted them. They are processed by `readDXF` but I haven't looked into whether they actually result in anything in the imported document.
- `3DFace` entities

furgo16, 2025-01-27:
I'm not sure if this helps, but ezdxf seems to be a in general a good reference beyond the official specs: https://ezdxf.readthedocs.io/en/stable/dxfinternals/tables/block_record_table.html