FreeCAD/FreeCAD #13600: DXF exporter wrongly scales arch radiuses

furgo16, 2024-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm importing a DXF file that I would like to scale from mm to m. To keep things simple, I've deleted everything from it except from the problematic area => [arch_radius_bug.zip](https://github.com/FreeCAD/FreeCAD/files/15076173/arch_radius_bug.zip)

If I try to import the file with these settings, it imports well:

[image]

[image]

FreeCAD document after import:
[arch_radius_bug_1.zip](https://github.com/FreeCAD/FreeCAD/files/15076187/arch_radius_bug_1.zip)

But as soon as I set the scale to 1000 in the import settings, the radiuses of the two arches are unproportionately scaled:

Zooming in:
[image]

Zooming out:
[image]

FreeCAD document after import:
[arch_radius_bug_1000.zip](https://github.com/FreeCAD/FreeCAD/files/15076198/arch_radius_bug_1000.zip)

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36961 (Git) Snap 996
Build type: Release
Branch: main
Hash: c4c453312d148fd67bd893366c42b559e28a85e6
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

File formats

### Anything else?

As a workaround, I would simply use the Draft_Scale command, but when I do, I'm presented with this error that prevents the scaling operation:

```
Unable to scale objects: Polyline, Polyline001, Line, Line001 - This object type cannot be scaled directly. Please use the clone method.
```

If I check the `Clone` option as suggested, scaling works, but the whole drawing is then one entity that I cannot easily manipulate (e.g. hide the individual items in layers).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-04-23:
The bulge of LWPoly segments is wrongly scaled. This should not happen as `Bulge=1/4*tan(arc angle)`.
https://github.com/FreeCAD/FreeCAD/blob/eb3e83b97a345620e0e26f3051d3d7a3f26755c6/src/Mod/Import/App/dxf/dxf.cpp#L2165
https://github.com/FreeCAD/FreeCAD/blob/eb3e83b97a345620e0e26f3051d3d7a3f26755c6/src/Mod/Import/App/dxf/dxf.cpp#L2211

Try the legacy DXF importer

furgo16, 2024-04-24:
> This should not happen

Thank you. I understand the `tan()` part of the bulge value, but I'm not sure I follow the code. Do you mean that the value of the bulge is (wrongly) multiplied by the scale factor?

> Try the legacy DXF importer

I did, and it works for this case, but it comes with its own set of alternate bugs when I run it over the original, bigger file that contains this sample segment to reproduce the bug. I've been filing a few of them, but it might take me a while and I'm not sure if it's worth it for the legacy importer (e.g. is it going to be phased out at some point?).

For now, I've been using the `ezdxf` library with good results, via the [`ezydxf` workbench](https://forum.freecad.org/viewtopic.php?t=81962).

PR #13678 (2024-04-29): Import: C++ DXF importer wrongly scaled polyline bulges
Fixes #13600.
commit c02df6a5 (2024-04-29): Import: C++ DXF importer wrongly scaled polyline bulges

Fixes #13600.