FreeCAD/FreeCAD #13605: [Problem] Unkown command(s), affected Draft functions

kenryuS, 2024-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I started up the CAD, I saw tons of logs saying "Unknown command '\<command name\>'". And I saw some error on top of those. Now I cannot use Draft functions.

<details>
<summary><h2>Full Log:</h2></summary>

```
22:57:04  During initialization the error "'PySide6.QtWidgets.QStatusBar.removeWidget' called with wrong argument types:
  PySide6.QtWidgets.QStatusBar.removeWidget(QPushButton)
Supported signatures:
  PySide6.QtWidgets.QStatusBar.removeWidget(PySide6.QtWidgets.QWidget)" occurred in /usr/lib/freecad/Mod/Tux/InitGui.py
22:57:04  Please look into the log file for further information
22:57:12  Path override failed for key base::DIR_APP_DICTIONARIES and path '/usr/lib/freecad/bin/qtwebengine_dictionaries'
22:57:12  shibokensupport/signature/parser.py:270: RuntimeWarning: pyside_type_init:_resolve_value

        UNRECOGNIZED:   'PySide6.QtGui.QPainterPath.Element'
        OFFENDING LINE: '0:PySide6.QtWidgets.QWidget.mapFrom(self,arg__1:PySide6.QtWidgets.QWidget,arg__2:typing.Union[PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element])->PySide6.QtCore.QPointF'
        
22:57:12  shibokensupport/signature/parser.py:359: RuntimeWarning: pyside_type_init:_resolve_type

                UNRECOGNIZED:   'typing.Union[PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element]'
                OFFENDING LINE: '0:PySide6.QtWidgets.QWidget.mapFrom(self,arg__1:PySide6.QtWidgets.QWidget,arg__2:typing.Union[PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element])->PySide6.QtCore.QPointF'
                
22:57:12  shibokensupport/signature/parser.py:270: RuntimeWarning: pyside_type_init:_resolve_value

        UNRECOGNIZED:   'PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element'
        OFFENDING LINE: '0:PySide6.QtWidgets.QWidget.mapFrom(self,arg__1:PySide6.QtWidgets.QWidget,arg__2:typing.Union[PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element])->PySide6.QtCore.QPointF'
        
22:57:12  shibokensupport/signature/parser.py:270: RuntimeWarning: pyside_type_init:_resolve_value

        UNRECOGNIZED:   'PySide6.QtGui.QPainterPath.Element'
        OFFENDING LINE: '0:PySide6.QtWidgets.QWidget.mapFromGlobal(self,arg__1:typing.Union[PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element])->PySide6.QtCore.QPointF'
        
22:57:12  shibokensupport/signature/parser.py:359: RuntimeWarning: pyside_type_init:_resolve_type

                UNRECOGNIZED:   'typing.Union[PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element]'
                OFFENDING LINE: '0:PySide6.QtWidgets.QWidget.mapFromGlobal(self,arg__1:typing.Union[PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element])->PySide6.QtCore.QPointF'
                
22:57:12  shibokensupport/signature/parser.py:270: RuntimeWarning: pyside_type_init:_resolve_value

        UNRECOGNIZED:   'PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element'
        OFFENDING LINE: '0:PySide6.QtWidgets.QWidget.mapFromGlobal(self,arg__1:typing.Union[PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element])->PySide6.QtCore.QPointF'
        
22:57:12  shibokensupport/signature/parser.py:270: RuntimeWarning: pyside_type_init:_resolve_value

        UNRECOGNIZED:   'PySide6.QtGui.QPainterPath.Element'
        OFFENDING LINE: '0:PySide6.QtWidgets.QWidget.mapFromParent(self,arg__1:typing.Union[PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element])->PySide6.QtCore.QPointF'
        
22:57:12  shibokensupport/signature/parser.py:359: RuntimeWarning: pyside_type_init:_resolve_type

                UNRECOGNIZED:   'typing.Union[PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element]'
                OFFENDING LINE: '0:PySide6.QtWidgets.QWidget.mapFromParent(self,arg__1:typing.Union[PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element])->PySide6.QtCore.QPointF'
                
22:57:12  shibokensupport/signature/parser.py:270: RuntimeWarning: pyside_type_init:_resolve_value

        UNRECOGNIZED:   'PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element'
        OFFENDING LINE: '0:PySide6.QtWidgets.QWidget.mapFromParent(self,arg__1:typing.Union[PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element])->PySide6.QtCore.QPointF'
        
22:57:12  shibokensupport/signature/parser.py:270: RuntimeWarning: pyside_type_init:_resolve_value

        UNRECOGNIZED:   'PySide6.QtGui.QPainterPath.Element'
        OFFENDING LINE: '0:PySide6.QtWidgets.QWidget.mapTo(self,arg__1:PySide6.QtWidgets.QWidget,arg__2:typing.Union[PySide6.QtCore.QPointF, PySide6.QtCore.QPoint, PySide6.QtGui.QPainterPath.Element])->PySide6.QtCore.QPointF'
        
22:57:12  shibokensupport/signature/parser.py:359: RuntimeWarning: pyside_type_init:_resolve_type

[... 192 lines cut ...]
```
</details>

### Full version info

```shell
[code]
OS: Arch Linux (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.8, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: C/Default (C) [ OS: Japanese/Japan (ja_JP) ]
[/code]
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-04-23:
This looks like the Arch Linux packaging team are getting ahead of themselves by using Qt 6.7.0 with FreeCAD 0.21.2.

luzpaz, 2024-04-23:
In another recent ticket there was a claim made that 21.2 can't be compiled with a a certain version of Qt6 and higher. Need citation for that.

djkato, 2024-04-23:
Hello, I +1 this. I couldn't build freecad-git from the AUR cause pyside2, pyside2-tools are now unresolvable depenencies. So I tried all day today to build freecad with QT6 and pyside6, which results in the Draft bench not opening anymore. 
`"NoneType" object has no attribute "subWindowActivated", /usr/lib/freecad/Mod/Draft/InitGui.py line 157, WorkingPlane.py line 1809`. inbetween all the unknown commands.

So AUR/freecad-git doesn't work cuz it doesn't build due to unresolvable deps,
Arch/Extra freecad doesn't work for cuz QT6 & pyside6 breaks Draft bench,
Building myself from source also doesn't work cuz same reason.
Webpack doesn't work cuz it crashes on some weird lib and segfaults.

I think it's time for me to go away from arch...

adrianinsaval, 2024-04-23:
you could always use the appimage or flatpak or install distrobox create an ubuntu image and install freecad or freecad-daily on it

djkato, 2024-04-24:
@adrianinsaval 
sorry, said webpack but I meant appimage.
```bash
/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
QOpenGLFunctions created with non-current context
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3c770) [0x721a2b25a770]
```
This is the error I get when trying to open the appimage from AUR/freecad-weekly-appimage
Today my last resort is distrobox, else I just give up.

YakoYakoYokuYoku, 2024-04-24:
I could replicate your problem, lucky for all of us #13611 fixes this. By the way, Arch merged a patch that I've sent that fixes the star imports which will help this issue and that should hit repos soon.

PR #13611 (2024-05-06): Fixups with Qt6 enums
Because PySide6 treats `enum`s and `QFlags` as `IntEnum`s and `IntFlag`s respectively I've added converters for Python PySide enums to C++, and due to this I've modified the requirements of task panel's `getStandardButtons` to return standard buttons without casting them to `int`. Fixes #13605 and part of #13303. Related to #6992.