FreeCAD/FreeCAD #13606: MultiTransform on PolarPattern → Traceback

mpfusion, 2024-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When a MultiTransform is applied to a PolarPattern then the following error appears:
````
pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'AttributeError'>: 'PartDesign.Feature' object has no attribute 'OriginalSubs'
````
And the model disappears. Forum discussion: https://forum.freecad.org/viewtopic.php?t=87096

[multi-transform.tar.gz](https://github.com/FreeCAD/FreeCAD/files/15078399/multi-transform.tar.gz)

### Full version info

```shell
FreeCAD 0.22.0, Libs: 0.22.0devR36963 (Git)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-04-23:
Please always make an effort to add the full About info. This extra minute of work can save a lot of hours of troubleshooting for others due to some  3rd-party dependency issue. TIA

HoWilgh, 2024-10-23:
can confirm this on:
`OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38641 +165 (Git) Snap 1162
Build type: Release
Branch: tag: 1.0rc2
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * MeshRemodel 1.9.18
  * fasteners 0.5.27
  * ThreadProfile 1.92.0
  * SelectionTools
  * Render 2023.12.28
  * Assembly3 0.12.2
  * kicadStepUpMod 10.19.4
`
The prior selected polar pattern is not automatically added and cant be added to the 'Added features'. To my understanding it should be automatically added into the 'Transformations'-Section in the Task-Dialogue.
Furthermore, when pressing 'Cancel' the Multi-Transform-Object stays in the Model-tree above the pattern but  should be removed.
The model zipped: [FC-Multitransform.zip](https://github.com/user-attachments/files/17487034/FC-Multitransform.zip)

[image]
[image]
[image]

FEA-eng, 2025-09-27:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```