FreeCAD/FreeCAD #13607: Continuous trim fails at a cluster of segments

prokoudine, 2024-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Enter the continuous trim mode, start deleting a cluster of segments. Occasionally, FreeCAD will throw 'Failed to trim edge' errors at you.

https://github.com/FreeCAD/FreeCAD/assets/57467/f6aaa627-16bd-4518-a068-aabf13ddd656

Example of an error message: 

```
17:45:47  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: Unknown C++ exception
17:45:47  Unnamed1#Sketch: App.getDocument('Unnamed1').getObject('Sketch').trim(16,App.Vector(-28.117377,-10.460073,0))
17:45:47  Unnamed1#Sketch: Failed to trim edge
```

Also when I click the OK button in the error message dialog and move the mouse pointer, the first segment the pointer crosses gets deleted (visible in the video above as well).

### Full version info

```shell
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu-xorg)
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.36972 (Git) AppImage
Build type: Release
Branch: flavor
Hash: fa03da43e8ad565b09a5ce11418eba8a141f4265
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.4.4
  * Lattice2 1.0.0
  * freecad.gears 1.2.0
  * OpenDark 2024.3.13
  * Ondsel-Lens 2024.1.22.01
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-03:
@prokoudine could you respond when you get a moment ?

FreeWilly2022, 2025-03-23:
I was able to recreate this error multiple times. The original commenter clicked very fast and I could not duplicate the exact movements. But selecting several lines in rapid succession, provided no constraint errors occurred, did result in this error. Here is one of the error reports:

```
12:06:53  Updating geometry: Error build geometry(6): Both points are equal
12:06:53  Invalid solution from SQP(augmented system) solver.
12:06:53  Sketch: Solving the sketch failed
12:06:53  Updating geometry: Error build geometry(6): Both points are equal
12:06:53  Invalid solution from SQP(augmented system) solver.
12:06:53  Sketch: Solving the sketch failed
12:06:53  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: Unknown C++ exception
12:06:53  Issue_13607#Sketch: App.getDocument('Unnamed').getObject('Sketch').trim(15,App.Vector(30.000000,61.927043,0))
12:06:53  Issue_13607#Sketch: Failed to trim edge
```

OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: arm64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09

closed 2025-07-09 by github-actions[bot]

prokoudine, 2025-07-12:
Did a quick test and I can't. The issue has been closed anyway, so no further action is expected :)