FreeCAD/FreeCAD #13608: [Problem] Setting the cache directory in the preferences is confusing

Roy-043, 2024-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD's cache directory can be set in the preferences:
https://wiki.freecad.org/Preferences_Editor#Cache

But the folder button on the right does not have that purpose. It is apparently intended for browsing the directory? To change the path you have type or copy-paste in the input box.

I suggest using the standard solution here. Compare the "Macro path" pref:
https://wiki.freecad.org/Preferences_Editor#Macro

Additionally the cache path has a mix of forward an backslashes on Windows:
`C:/Users/name/AppData/Local/cache\FreeCAD\Cache\`

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git)
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-04-26:
Actually the cache directory is read-only. Will do a PR to add "read-only" to the "Location" label. Will also add a tool tip to the Browse button.

Issue for the mixed path separators: #11970 (reopened).

closed 2024-04-29 by sliptonic

vertexgamer, 2024-05-29:
Still an issue

PR #13666 (2024-04-29): Gui: Clarify cache directory as read-only in prefs, add Browse tooltip
Fixes #13608.
commit dd882862 (2024-04-29): Gui: Clarify cache directory as read-only in prefs, add Browse tooltip

Fixes #13608.