FreeCAD/FreeCAD #13610: TabBar highlights the wrong workbenches

prokoudine, 2024-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With TabBar workbench selector enabled, when I switch from PD to Sketcher by creating a new sketch, Sheet Metal gets highlighted instead of Sketcher in the TabBar. When I go back to PD by closing the sketch in the Task panel, FEM gets highlighted instead of PD. Demo:

https://github.com/FreeCAD/FreeCAD/assets/57467/6726bb3b-59b4-466f-a4f8-2c815e2d4e6c

I can reproduce this with any UI theme that I have around.

I cannot reproduce this with the ComboBox workbench selector.

### Full version info

```shell
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git) AppImage
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 2.9.0
  * DynamicData 2.60.0
  * Assembly4 0.50.6
  * ExplodedAssembly
  * sheetmetal 0.3.15
  * BIM 2021.12.0
  * A2plus 0.4.61
  * OpenTheme 2024.4.20
  * Render 2023.12.28
  * Rocket 3.2.4
  * Curves 0.6.21
  * freecad.gears 1.2.0
  * Assembly3 0.12.0
  * boltsfc 2022.11.5
  * OpenDark 2023.12.17
  * dodo 1.0.0
  * Ondsel-Lens 2023.12.13.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-23:
Can confirm on Windows.

kadet1090, 2024-05-03:
@prokoudine Is this still relevant? I don't know If I should add this to fixlist of #13721 or maybe it was fixed before in the https://github.com/FreeCAD/FreeCAD/pull/13624

FEA-eng, 2024-06-29:
I can't reproduce this issue.

```
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git)
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.4.20
  * freecad.gears 1.2.0
```

maxwxyz, 2024-06-29:
@prokoudine is this fixed on your system as well? I still cannot reproduce this.

prokoudine, 2024-07-01:
Seems like it was fixed in one of commits

closed 2024-07-01 by prokoudine