FreeCAD/FreeCAD #13613: both freecad stable and weekly stopped loading gui after error in freecad weekly builds

zamonary1, 2024-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was playing with new features in freecad weekly when suddenly my gui has become fully transparent and flickery. I thought that was because of unstable build. I saved my project but after restarting freecad it failed to load gui. What's interesting is that the problem remains in freecad stable as well, after restart too. There wasn't any system updates at the background.

https://github.com/FreeCAD/FreeCAD/assets/73584226/39f2cc37-70b7-4f29-b014-e7ed2b0b0388
[image]

### Full version info

```shell
[code]
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * Curves 0.6.13
  * ThreadProfile 1.89.0
  * LCInterlocking
  * sheetmetal 0.3.1
  * CadbaseLibrary 0.3.0
  * retr3d
  * 3D_Printing_Tools
  * fasteners 0.4.66
  * InventorLoader 1.4.0
  * A2plus 0.4.61
  * freecad.gears 1.0.0
  * Design456 0.0.1
  * AnimationFreeCAD 1.0.0
  * Render 2024.1.23
  * CurvedShapes 1.0.5

[/code]
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git) AppImage
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * Curves 0.6.13
  * ThreadProfile 1.89.0
  * LCInterlocking
  * sheetmetal 0.3.1
  * CadbaseLibrary 0.3.0
  * retr3d
  * 3D_Printing_Tools
  * fasteners 0.4.66
  * InventorLoader 1.4.0
  * A2plus 0.4.61
  * freecad.gears 1.0.0
  * Design456 0.0.1
  * AnimationFreeCAD 1.0.0
  * Render 2024.1.23
  * CurvedShapes 1.0.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-24:
could you try this with a fresh install and removed mods/config files?

komoricodrutz, 2024-04-24:
Hi all,
It seems to be more the case that the "Start Page" has disappeared from the workbenches list.
If you start the stable version and set the "Start Page" to open automatically when you start, then restart the stable version, it will be fine.
However, if after that you start the Daily version, then it's blank and the "Start" workbench has disappeared and cannot be selected.
The user.cfg also changes the AutoLoadModule to PartDesignworkbench
```
<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<FCParameters>
  <FCParamGroup Name="Root">
    <FCParamGroup Name="BaseApp">
      <FCParamGroup Name="LogLevels">
        <FCInt Name="Default" Value="2"/>
      </FCParamGroup>
      <FCParamGroup Name="Preferences">
        <FCParamGroup Name="General">
          <FCText Name="FileOpenSavePath">/mnt/hdd/data/My Crap/DIY Projects/3DPrn/Mine/Shields</FCText>
          <FCText Name="LastModule">StartWorkbench</FCText>
          <FCText Name="AutoloadModule">PartDesignWorkbench</FCText>
```

Build:
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36961 (Git)
Build type: Release
Branch: main
Hash: c4c453312d148fd67bd893366c42b559e28a85e6
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BOLTSFC 2022.11.5
  * A2plus 0.4.64a
  * Lithophane
  * Assembly4 0.50.12
  * Glass
  * fasteners 0.5.17
  * fcgear 1.0.0

maxwxyz, 2024-04-24:
@chennes FYI

komoricodrutz, 2024-04-24:
I just tested opening a file from File -> Open Recent and then playing around in it in the Part Design workbench and seems to be working just fine.
When starting it from Terminal, there doesn't seem to be a related error:
[image]

The error there about the icon is in both versions and let's say it's not necessarily a dealbreaker.

Rexbas, 2024-04-24:
Start has been changed in #13134 but a migration script did not work properly. The migration script is fixed in #13609 but is not included in the weekly build right now. If you don't see Start in the weekly build then you can set `ShowOnStartup` to true using Tools->Edit parameters... at BaseApp/Preferences/Mod/Start.

If you go from weekly build to stable you may indeed have to manually select the Start workbench as the startup workbench.

zamonary1, 2024-04-24:
> you can set `ShowOnStartup` to true using Tools->Edit parameters... at BaseApp/Preferences/Mod/Start.
> 
> If you go from weekly build to stable you may indeed have to manually select the Start workbench as the startup workbench.

Yeah, now i can open freecad and projects but menus are still transparent and glitchy.
[image]
I am not sure how to set the default workbench in freecad stable, it is still broken

zamonary1, 2024-04-24:
Okay, i've got freecad stable to work by loading start workbench from the preferences.
[image]
[image]

komoricodrutz, 2024-04-24:
Also found it by accident in the HELP menu. However it is severely "mutilated". 
Just updated to:
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36967 (Git)
Build type: Release
Branch: main
Hash: 24657494c9b142e688c1910ecb34083f88ef78be
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BOLTSFC 2022.11.5
  * A2plus 0.4.64a
  * Lithophane
  * Assembly4 0.50.12
  * Glass
  * fasteners 0.5.17
  * fcgear 1.0.0

Mutilation screenshot:
[image]

But I guess, as it's still a work in progress, it's fine.

luzpaz, 2024-08-30:
Please retest on latest dev. If possible, backup config files and move them so the can be recreatred.

luzpaz, 2024-10-24:
Is this still relevant ?

maxwxyz, 2024-11-09:
@zamonary1 please test on the latest weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
If there is still an issue please comment and reopen the issue.

closed 2024-11-09 by maxwxyz