FreeCAD/FreeCAD #13615: TechDraw: Update dimension tolerance style to ISO 129-1:2022

maxwxyz, 2024-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Current implementation: 
[image]

Correct implementation acc. to **ISO129-1:2022**:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git)
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.9
  * Curves 0.6.35
  * fasteners 0.5.17
  * OpenTheme 2024.4.20
  * OpticsWorkbench 1.0.16
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-04-24:
This sounds familiar.  fca3b80 addressed some of this, I think.

maxwxyz, 2024-04-24:
Some parts, yes but the preview in OP is from the current weekly build which includes this commit.

pierreporte, 2024-04-24:
There should be a setting to switch between tolerancing style. ISO-129 style may be standard but the current one is widely used.