FreeCAD/FreeCAD #13616: TechDraw: Support Hole Dimensions (counterbore, countersunk) ISO 129-1:2022

maxwxyz, 2024-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

TechDraw should support symbols for dimensioning holes according to ISO129-1:2022.
These include, hole depth, countersunk, counterbore and diameter:
[image]

Symbols according to dimension text height:
[image]

Positioning and relation to text:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git)
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.9
  * Curves 0.6.35
  * fasteners 0.5.17
  * OpenTheme 2024.4.20
  * OpticsWorkbench 1.0.16
  * sheetmetal 0.4.11
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-24:
@WandererFan currently most symbols are already included in the ExtensionCustomizeFormat but I've found no way to type text inside the symbols or use them as described above in two lines (with line break) on the dimension line

eduncan911, 2025-02-17:
I think this can be closed now.  

I've been manually entering these for a few years in FreeCAD TechDrawings as normal diameter dimensions (editing the properties).

However, we now have a new tool in the TechDraw WB since FreeCAD 1.0 called "Customized Format Label", and it has most of the common symbols readily available.

To use:

1. make a normal diameter demision
2. select the new demionsion you just did, and activate the "Customized Format Label"
3. a new dialog Task is shown, with several options.

[image]

cc @WandererFan @pierreporte

maxwxyz, 2025-02-18:
Ideally, TD knows what hole this is and dimensions them automatically

pierreporte, 2025-02-18:
Basically, the custom format task panel would provide several value placeholder that correspond to the various sub-features of the hole so that they could be rearranged freely while being able to be updated automatically when geometry changes. While it is indeed possible to use the symbols, the actual dimensions need to be entered by hand.

eduncan911, 2025-02-19:
> Ideally, TD knows what hole this is and dimensions them automatically

You speak or a totally different feautr called Automatic Dimensioning...  Is there a tool in the TD WB I am unaware of that does this already?

If you are wanting an automatic dimensioning tool, Id advise to start a new issue because I don't think it exists today and it would have to cover far more than just counterbores.

/TL;DR

I've used SolidWorks' "automatic" dimensioning tool over the years and I can tell you, it's not all that good.  Not all all.  it produces very sloppy, and often incomplete diagrams.  But because of the sloppiness, it's hard to sit in and figure out what areas it just "skipped" because it doesn't know what to do.

I still do manual dimensions in SW as it's the only way to be sure it is all covered.  Especially when dealing with overseas vendors, it's better to do it right the first time.

pierreporte, 2025-02-19:
@eduncan911 This is not automatic dimensioning but something similar to a feature Solidworks have: https://help.solidworks.com/2021/english/SolidWorks/sldworks/c_modify_hole_callout_format.htm

pierreporte, 2025-04-29:
A **150 € reward** will be given to the contributor that writes a pull request, solving this issue, that gets merged. It will be paid by the FreeCAD Project Association. It is to be considered an incentive. Grant applications are welcome, but the total amount will exclude the reward.