FreeCAD/FreeCAD #13618: Some actions cannot be undone

mpfusion, 2024-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Some actions cannot be undone using ”undo” or ctrl-z. Examples are
- horizontal/vertical extent dimensions
- hole/shaft fits
- probabaly a few others as well, not sure

Steps to  reproduce:

1st example
- create simple TechDrawing or open attached file
- open TD workbench
  - select circle
  - insert horizontal extent dimension
  → dimension is created, but undo is greyed out

2nd example
  - select circle
  - insert diameter dimension
  - select face
  - add centerline to faces → OK
  - select diameter dimension
  - add hole or shaft fit → shaft fit, c11  → OK
  - undo
  → the centerline disappears instead of the shaft fit, which was the last action

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36963 (Git)
Build type: Release
Branch: main
Hash: 68bfbd0707f96be2bd5a6bf41574eff089d29dd7
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Installed mods: 
  * BOLTSFC
  * kicadStepUpMod 11.1.2
```

### Subproject(s) affected?

Techdraw

### Anything else?

[cosmetic-undo.tar.gz](https://github.com/FreeCAD/FreeCAD/files/15090596/cosmetic-undo.tar.gz)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-25:
Undo/Redo bugs need to be prioritized. It's very demoralizing when they don't work correctly. They are 'Workflow Killers'.

JohnOCFII, 2025-01-25:
And, while you may wish to separate out issues by workbench, there are many other areas in FreeCAD where Undo/Redo don't work, such as moving task panels/tabs to different locations in the UI.

[image]

luzpaz, 2025-01-25:
Indeed. We really need to tighten up on that issue. Undo/Redo bugs really suck.

dannystaple, 2025-02-28:
Given how unexpected it is when the wrong item is undone, or access violations appear, this sounds more like a bug than a feature request. Not only does it not always undo what you expected, but you are now questioning the integrity of the current document, making it safer to close the document unsaved and go back to a last save point after seeing the wrong item undone (or not seeing what it did undo).

psommerfeld, 2025-04-12:
I agree, this is a Bug, not a Feature!

Just encountered these types of issues in TechDraw.
Undoing a move/delete results in one or more more distant actions being undone.
ie. Attached video of attached project shows undo of moving a view restores previously deleted annotations among other things.

[techdraw undo view move fails.FCStd.zip](https://github.com/user-attachments/files/19719546/techdraw.undo.view.move.fails.FCStd.zip)

https://github.com/user-attachments/assets/d527af2f-57ba-47d1-a987-68d9ab702d57

luzpaz, 2025-05-07:
@maxwxyz undo/redo ticket is a feature ?

maxwxyz, 2025-05-07:
if it is currently not implemented it sounds like a new feature which needs to be coded. If it is implemented but does not work I would consider it a bug.

luzpaz, 2025-05-07:
@maxwxyz the undo/redo commands exist and users are using them so technically this could be a bug as well.

maxwxyz, 2025-05-07:
Yes from a UI point of view I agree. But undo/redo is also not supported in many other areas in FreeCAD unfortunately. We can label this as bug but I guess that won't change anything 😄

JohnOCFII, 2025-05-07:
There are currently (at least 23) open issues referencing Undo.  I'd guess to properly attack this, it would need a broader discussion among the workbench maintainers and developers  to determine the best way forward.  Not even sure if this is a global feature across FreeCAD, or if it is implemented Workbench by Workbench. 

https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aissue%20state%3Aopen%20in%3Atitle%20Undo%20