FreeCAD/FreeCAD #13619: Exporting a drawing marks the file as modified

mpfusion, 2024-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Exporting a drawing marks the file as modified. I expect no modifications from merely exporting a drawing.

Steps to reproduce:

- open a TechDrawing
- right-click → Export PDF
- “foo.pdf” → Save
→ file is marked modified (asterisk)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36963 (Git)
Build type: Release
Branch: main
Hash: 68bfbd0707f96be2bd5a6bf41574eff089d29dd7
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Installed mods: 
  * BOLTSFC
  * kicadStepUpMod 11.1.2
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-06-10:
The ShowFrames property of ViewProviderPage is modified as part of the printing process.   This marks the document as modified.

mpfusion, 2024-06-11:
That explains *why* it happens, but it's not what the user expects. Exporting a file shouldn't modify the file. And in fact it's not, then don't mark it as such.

Maybe a better approach is to store the old ShowFrames value as well as the “modified” state, modify ShowFrames if desired, then restore the original value of ShowFrames and reset the “modified” state to what it was before exporting. I know nothing about the code, so that might not work.

luzpaz, 2024-10-20:
@WandererFan :point_up_2:

WandererFan, 2024-10-30:
Should be fixed by PR #17488.

luzpaz, 2024-10-30:
@mpfusion can you test and report back ? TIA

mpfusion, 2024-10-30:
It appears to be fixed on https://github.com/FreeCAD/FreeCAD/commit/b6cda4309a9031bebffa73b694e9a9cc125d64de. Thanks for the fix. Your work in much appreciated.

closed 2024-10-30 by luzpaz