FreeCAD/FreeCAD #13620: DXF importer imports file when clicking on the Cancel button

furgo16, 2024-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
1. These steps assume the "Show this dialog when importing and exporting" under Preferences > Import-Export >DXF is checked
2. Go to File > Import...
3. Select a DXF file. The DXF import/export window opens
4. Click on Cancel
5. Expected: the import/export window closes, no further action is taken
6. Actual: the import/export window closes, the DXF file is imported and the document is populated with the imported entities 

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36967 (Git) Snap 997
Build type: Release
Branch: main
Hash: 24657494c9b142e688c1910ecb34083f88ef78be
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * ezydxf
  * Assembly4 0.50.12
  * ExplodedAssembly
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-04-25:
The Cancel button is only related to the Preferences editor.

yorikvanhavre, 2024-04-29:
Ideally, the dialog that pops up when the dxf importer starts, would not be the preferences editor itself, but a different dialog tailored for that purpose only...

furgo16, 2024-04-29:
[image]

> The Cancel button is only related to the Preferences editor.

Indeed, but regardless it comes as a surprise to the user that clicking on the <kbd>Cancel</kbd> that there is an action taken at all.

It took me a while to realize this was happening, as I was testing different import settings and importers to correctly import my DXF files. It only dawned on me when I saw that in one of my files the DXF had been imported twice.

> Ideally, the dialog that pops up when the dxf importer starts, would not be the preferences editor itself, but a different dialog tailored for that purpose only...

That was my thought as well. I'll file a separate issue.

furgo16, 2024-05-01:
Filed https://github.com/FreeCAD/FreeCAD/issues/13751 as a follow up.

furgo16, 2025-06-25:
Still an issue. I might actually look into it soon.

closed 2025-07-28 by yorikvanhavre