FreeCAD/FreeCAD #13623: Path tool /Output File no systems like G54 showed, only empty boxes

reset12, 2024-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After open Pathtool and create a Job, in the Output tab there is no more any description after the clickboxes of systems (like G53,G54, etc).
At the start of my work the description was available, but after a crash it was absent.

[image]

### Full version info

```shell
[code]
OS: Debian GNU/Linux 11 (bullseye) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * ThreadProfile 1.89.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-24:
could you test this in the current dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

reset12, 2024-04-24:
I have the error not in the weekly build, but now also no more in the stable version. I can't reproduce the error now. 
I have errors in the console-window, can this a part of the problem ?
`22:27:41  Migrating Start Workbench to Start command... done.
22:28:22  Traceback (most recent call last):
22:28:22    File "/tmp/.mount_FreeCAXcT4tC/usr/Mod/CAM/Path/Main/Gui/Job.py", line 167, in setEdit
22:28:22      self.openTaskPanel()
22:28:22    File "/tmp/.mount_FreeCAXcT4tC/usr/Mod/CAM/Path/Main/Gui/Job.py", line 174, in openTaskPanel
22:28:22      self.taskPanel.setupUi(activate)
22:28:22    File "/tmp/.mount_FreeCAXcT4tC/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1489, in setupUi
22:28:22      self.setFields()
22:28:22    File "/tmp/.mount_FreeCAXcT4tC/usr/Mod/CAM/Path/Main/Gui/Job.py", line 911, in setFields
22:28:22      item = self.form.wcslist.findItems(f, QtCore.Qt.MatchExactly)[0]
22:28:22             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
22:28:22  IndexError: list index out of range

`
[source.zip](https://github.com/FreeCAD/FreeCAD/files/15100092/source.zip)

maxwxyz, 2024-04-26:
I'll close this then if not reproducible. You can comment if this appears again.

closed 2024-04-26 by maxwxyz