FreeCAD/FreeCAD #13628: Right-clicking a document in the tree view activates it directly and folds other items before executing the selected command.

farleyrunkel, 2024-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Based on the user's intuition, only the right-click menu will appear When a right-click is made.
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36916 (Git)
Build type: Release
Branch: refractor_tracker
Hash: 31f5ca51dae6ae11343e639eb253b6e31f608379
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Chinese/China (zh_CN)
Installed mods: 
  * nurbs
```

### Subproject(s) affected?

Core

### Anything else?

Suggesstion:
- Disable document activation when right-click
- Add a Qt-like command "Set current document as active document" in right-click menu?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-04-26:
Related: https://wiki.freecad.org/Std_TreeMultiDocument.

farleyrunkel, 2024-04-26:
Thanks for your reply. This is a minor issue, but it seems not necessary to activate a document when right-clicking it.

wwmayer, 2024-05-07:
Can this be closed?

farleyrunkel, 2024-05-07:
sure

closed 2024-05-07 by farleyrunkel