FreeCAD/FreeCAD #13629: Sketcher: Weird behaviour when adding distance between arc and line

wwmayer, 2024-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

See German sub-forum https://forum.freecad.org/viewtopic.php?t=87119

Apparently the issue is caused by https://github.com/FreeCAD/FreeCAD/pull/12012

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36976 (Git)
Build type: debug
Branch: main
Hash: a389403cdc4019b40622627926573b1613284582
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-04-24:
Create a sketch with this code snippet:
```
doc = App.newDocument()
body = doc.addObject('PartDesign::Body','Body')
sketch = body.newObject('Sketcher::SketchObject','Sketch')
sketch.AttachmentSupport = (doc.getObject('XY_Plane'),[''])
sketch.MapMode = 'FlatFace'

geoList = []
geoList.append(Part.ArcOfCircle(Part.Circle(App.Vector(-66.131020, -10.357095, 0.000000), App.Vector(0.000000, 0.000000, 1.000000), 3.409007), 4.279981, 7.507065))
geoList.append(Part.LineSegment(App.Vector(-76.131020, 7.380997, 0.000000), App.Vector(-77.321495, -31.666620, 0.000000)))
sketch.addGeometry(geoList, False)
del geoList

sketch.addConstraint(Sketcher.Constraint('Vertical', 1))
sketch.addConstraint(Sketcher.Constraint('Distance',0,1,1,8.571427))
sketch.setDatum(1,App.Units.Quantity('8.571430 mm'))
doc.recompute()
```

Edit the sketch and do a fit all if needed. Trying to move the datum constraint causes totally un-intuitive behaviour. Moving the mouse to up or down moves the label to left or right. Moving the mouse to right or light moves the label to bottom or top.

wwmayer, 2024-04-24:
Second example:
```
doc = App.newDocument()
body = doc.addObject('PartDesign::Body','Body')
sketch = body.newObject('Sketcher::SketchObject','Sketch')
sketch.AttachmentSupport = (doc.getObject('XY_Plane'),[''])
sketch.MapMode = 'FlatFace'

geoList = []
geoList.append(Part.ArcOfCircle(Part.Circle(App.Vector(-44.226295, 6.428696, 0.000000), App.Vector(0.000000, 0.000000, 1.000000), 21.102344), 1.058407, 3.648143))
geoList.append(Part.ArcOfCircle(Part.Circle(App.Vector(-26.250109, -12.261777, 0.000000), App.Vector(0.000000, 0.000000, 1.000000), 7.622766), 4.037648, 6.851029))
sketch.addGeometry(geoList,False)
del geoList

sketch.addConstraint(Sketcher.Constraint('Distance',0,3,1,18.309396))
sketch.setDatum(0,App.Units.Quantity('18.309400 mm'))
doc.recompute()
```

This adds a distance constraint between the center point of the big circle and the shape of the small circle. Showing the helper circle for the small one is OK but I don't think the helper circle of the big one is anyhow helpful because the constraint reference a point of this circle and not its shape.

PaddleStroke, 2024-05-17:
@edi271 yes, if I'm not tagged I'm not going to see it.

I see 2 problems here.
- The arc helper should appear only if constraining the edge.
- The constraint is moving weirdly in your first example.

PaddleStroke, 2024-05-17:
Second bug happen specifically when you make a distance between a line edge and an arc end point. @FlachyJoe does that ring a bell to you?

FlachyJoe, 2024-05-17:
@PaddleStroke it can be an ordering problem of conditions. We have a big stack of `if else` and it's easy to catch a case in a wrong one.

PaddleStroke, 2024-05-17:
Ok I'm on it, its in VPSketch moveConstraint.

PaddleStroke, 2024-05-17:
https://github.com/FreeCAD/FreeCAD/pull/14099

luzpaz, 2024-08-11:
@PaddleStroke is this issue still relevant ?

PaddleStroke, 2024-08-11:
That's fixex

closed 2025-07-09 by github-actions[bot]