FreeCAD/FreeCAD #13630: TabBar wrongly activates always the last tab when loading a new workbench

furgo16, 2024-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. With the TabBar enabled¹, click on the `+` (plus sign) at the beginning of the workbenches tabs.
2. Choose a workbench from the context menu shown
3. **Actual**: the workbench is loaded (without a tab as per issue #13602. The tab of the last workbench in the existing tabs is activated (in my case, the Assembly tab)
4. **Expected**: the workbench is loaded (with or without a tab is a pending design decision). The tab of the last workbench should not be arbitrarily activated.

[1] Edit > Preferences... > Available Workbenches > Workbench selector type => TabBar

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36961 (Git) Snap 996
Build type: Release
Branch: main
Hash: c4c453312d148fd67bd893366c42b559e28a85e6
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-24:
So you mean: Just display the current WB on a fresh start and add new tabs if you want to change a WB?

furgo16, 2024-04-24:
Not exactly (unless I haven't understood it correctly)

Currently:
- The tabs list contains only the auto-loaded workbenches
- The `+` sign opens a list of the rest of installed workbenches
- Choosing a workbench from the list loads it, it displays it, but it does not create a tab for it

This proposal:
- The tabs list contains only the auto-loaded workbenches
- The `+` sign opens a list of the rest of installed workbenches
- Choosing a workbench from the list loads it, displays it, and creates a tab for it. If the list of tabs exceeds the list's fixed width, the scrolling arrows are shown.

At the end of the day, this issue is about not having orphaned workbenches without tabs.

I hope this clarifies things.

maxwxyz, 2024-04-24:
Why not use it like a tab in the file browser / terminal / web browser?

- Just load the defined WB which is set as default/start WB.
- If you create a sketch or double click a sketch or spreadsheet it automatically opens/adds the Sketcher WB (like you can open a link in a new tab in a web browser).
- You can manually open tabs with the `+` and choose from all active but not opened WB and these stay open as tabs.

furgo16, 2024-04-24:
That covers the essence of the proposal, which is not having tab-less workbenches, so I'd be fine with that too. A quick note:

- I still think it's useful to have a set of auto-loaded workbenches that are opened as tabs at startup. It's useful for quick access: you cannot always double click on an element and open the desired workbench. E.g. if I'm on the BIM workbench and I need to convert wires to a face to build a wall, I'll quickly switch to Part > Make face from wires.

furgo16, 2024-04-24:
Trying to mention @FreeCAD/design-working-group for input does not seem to work, although it's the convention described at https://github.blog/2012-05-09-introducing-team-mentions. Does that team already exist? Is it mentionable only by members of the FreeCad organization on GitHub?

PR #13721 (2024-05-13): Gui: Move more button to the end WB TabBar
Please don't squash - I keep the commit history meaningful :)

This changes back placement of the "more" button of the WB TabBar to be at the end, where it should be naturally placed. In order to ensure that it is always visible the control was reworked to show this button always after the tab bar widget which now is dynamically sized. This is behavior that is well known from browsers.

[image]

This commit also ensures that active workbench is always visible in the TabBar by adding additional temporary tab when necessary. This tab will automatically  [... cut]
commit 3405939f (2024-05-13): Gui: Move more button to the end WB TabBar

This changes back placement of the "more" button of the WB TabBar to be
at the end, where it should be naturally placed. In order to ensure that
it is always visible the control was reworked to show this button always
after the tab bar widget which now is dynamically sized. This is
behavior that is well known from browsers.

This commit also ensures that active workbench is always visible in the
TabBar by adding additional temporary tab when necessary. This tab will
automatically dissapear when not needed.

Fixes: #13720
Fixes: #13630
closed 2024-05-13 by chennes