FreeCAD/FreeCAD #13632: Addon Manager: Preview image for addons

maxwxyz, 2024-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently addons in the addon manager are displayed with a list containing the icon and the name, a click on them reveals the readme.md info for that addon. Most addons require a restart, so there is no quick way to preview a new WB, the functionality of a macro or the look of a theme.
The readme info is very individual, some have text only, some have images, instructions how to install or use it and documentation or help info.
It would be great to have the ability for each 3rd party addon to provide a dedicated preview image, which could be displayed in the addon manager to provide a quick preview of what the addon looks like or what it is intended for.

Regarding themes or workbenches it would be great to have a bigger preview image or gallery, maybe on the new start page or for the first-run wizard so users can get a look before installing and restarting and I think this could be a more unified appearance.

Maybe guidelines and default sizes for addon previews could be provided as a prerequisite.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-24:
@MisterMakerNL @obelisk79 please add your feedback

obelisk79, 2024-04-24:
Max, these seem like relatively decent ideas. But I think more importantly, it might be a good idea to create templates for add-on information so they are presented in a common format. Regarding a gallery view for themes, this is probably a good idea.

hasecilu, 2024-09-18:
Ideally the README should be enough but since some WB have so many features putting images of of all would bloat it.

The `package.xml` file already contains a line to set the WB icon, something similar could be added to the file:

```xml
  <icon>freecad/freegrid/resources/icons/FreeGrid.svg</icon>
  <gallery>img/gallery/Box.jpg</gallery>
  <gallery>img/gallery/Grid.jpg</gallery>
  <gallery>img/gallery/BitCartridge.jpg</gallery>
  <gallery>img/gallery/AllInOne.jpg</gallery>
```

Since the Addon Manager dialog usually is small clicking an `Open gallery` button opens another dialog on fullscreen/big size (ideally Window Managers don't mess with size) with images with 16:9 aspect ratio.