FreeCAD/FreeCAD #13636: Children objects become disconnected from their parent layers

furgo16, 2024-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've created a test file with 3 layers: 2 of them behave as expected when dragging and dropping elements in and out (the children change style as defined by the layer) or changing the parent layer's visibility (when off, none of the children are visibly, and viceversa). However, one of them remains somehow disconnected when either creating or dragging and dropping children: styles are not applied and changing visibility of the layer does not affect the visibility of the children.

I believe I created all layers in the same way, so I'm uncertain of what triggered the behavior for `Layer000`. I first became aware of this issue on a FreeCAD document into which I imported a DXF file: I tried toggling visibility of the layers but it would not affect their children.

The attached file serves to reproduce that, but unfortunately I cannot pinpoint what triggers it in the first place.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36961 (Git) Snap 996
Build type: Release
Branch: main
Hash: c4c453312d148fd67bd893366c42b559e28a85e6
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

Draft

### Anything else?

[layer-visibility.zip](https://github.com/FreeCAD/FreeCAD/files/15100474/layer-visibility.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-25:
@Roy-043 is this regarding your PR https://github.com/FreeCAD/FreeCAD/pull/13633

Roy-043, 2024-04-25:
Yes that is correct. Layer000 has an `OverrideShapeAppearanceChildren` property instead of an `OverrideShapeColorChildren` property. The other layers do not have this, My guess would be that they have been created in a previous version but that does not necessarily tally with:

> I believe I created all layers in the same way

furgo16, 2024-04-25:
I did quite a lot of tests and comparison with FreeCAD 0.21 until I understood what the issue was. Reading this, I cannot discard that I mistakenly created `Layer000` with 0.21 instead of 0.22.

Is there a temporary workaround to be able to use layers properly in 0.22 until https://github.com/FreeCAD/FreeCAD/pull/13633 is merged? E.g. can properties be created dynamically to add the `OverrideShapeColorChildren` property layers are expecting?

Roy-043, 2024-04-26:
Fixed with #13633.

closed 2024-04-26 by Roy-043