FreeCAD/FreeCAD #13637: Crash (SIGSEGV) when clicking outside of FreeCAD window when a contextual menu is open

yomgui1, 2024-04-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD segfault when a left mouse click is done outside of the FreeCAD window when a contextual menu is open.

```FreeCAD 0.22.0, Libs: 0.22.0devR36958 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3dbb0) [0x7f98fba5fbb0]
#1  /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libFreeCADGui.so(+0x8d63e9) [0x7f98fe6d63e9]
#2  /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libFreeCADGui.so(+0x8da349) [0x7f98fe6da349]
#3  0x7f98fe6d8632 in Gui::GestureNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libFreeCADGui.so+0x2a2
#4  0x7f98fe6cb2d5 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libFreeCADGui.so+0x65
#5  0x7f98fe6f3ccb in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libFreeCADGui.so+0x6b
#6  0x7f98fe6aa355 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libFreeCADGui.so+0x1e5
#7  0x7f98fb6b68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Core.so.5+0x7b
#8  0x7f98fc57039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#9  0x7f98fc5778f9 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Widgets.so.5+0xa39
#10  0x7f98fe326ff9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libFreeCADGui.so+0x99
#11  0x7f98fb6b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Core.so.5+0x108
#12  0x7f98fc5763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#13  /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7f98fc5c6e6f]
#14  /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7f98fc5ca0c9]
#15  0x7f98fc5703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#16  0x7f98fe326ff9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libFreeCADGui.so+0x99
#17  0x7f98fb6b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Core.so.5+0x108
#18  0x7f98fbd353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Gui.so.5+0x84c
#19  0x7f98fbd0b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Gui.so.5+0xac
#20  /tmp/.mount_freecagmrPZ4/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7f98e8de0bea]
#21  /tmp/.mount_freecagmrPZ4/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7f98f9f10ad7]
#22  /tmp/.mount_freecagmrPZ4/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7f98f9f13ac7]
#23  /tmp/.mount_freecagmrPZ4/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f98f9f140ce]
#24  0x7f98fb70aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Core.so.5+0x66
#25  0x7f98fb6b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Core.so.5+0x11b
#26  0x7f98fb6bd212 in QCoreApplication::exec() from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libQt5Core.so.5+0x82
#27  0x7f98fe28a587 in Gui::Application::runApplication() from /tmp/.mount_freecagmrPZ4/usr/bin/../lib/libFreeCADGui.so+0xa27
#28  /tmp/.mount_freecagmrPZ4/usr/bin/freecad(+0x7620) [0x56247367f620]
#29  /lib64/libc.so.6(+0x27b8a) [0x7f98fba49b8a]
#30  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7f98fba49c4b]
#31  /tmp/.mount_freecagmrPZ4/usr/bin/freecad(+0x7981) [0x56247367f981]
```

How to reproduce:
- run FreeCAD
- create a new document
- go on Part WB
- note: I have enabled the new UI system with transparent Model/Task areas on left.
- play with the Task/Model subwindows (move/drag/resize)
- right click on the empty 3D space to open the contextual menu
- move the mouse cursor on another screen, outside of FreeCAD main window (I have two physical screen) then left click
- the crash occurs

### Full version info

```shell
OS: Fedora Linux 38 (Workstation Edition) (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git) AppImage
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: French/France (fr_FR) ]
Installed mods: 
  * Render 2024.1.23
  * BillOfMaterials 0.0.7.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-25:
I cannot reproduce on Windows.

yomgui1, 2024-04-25:
Maybe. I only test on GNU/Linux (Fedora 38 here)

xtemp09, 2024-04-25:
I cannot reproduce on 
<details><summary>
(1)
</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36882 (Git)
Build type: Debug
Branch: main
Hash: cf1eeaab19da2d2a6eaaba3ae6bd1fc8ad620552
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

<details><summary>
(2)
</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git) AppImage
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

</details>

@yomgui1, could you record a video to show the exact sequence of actions? "play with the Task/Model subwindows (move/drag/resize)" is too vague.

yomgui1, 2024-04-25:
@xtemp09 

For sure!

Uploading 2024-04-25 20-35-13.mp4…

yomgui1, 2024-04-25:
@xtemp09 

https://github.com/FreeCAD/FreeCAD/assets/8989517/0c655fbb-ffda-48e1-83ff-11fa0a23f9b9

xtemp09, 2024-05-05:
I've just downloaded Fedora 40 and the latest weekly bundle. No crash occurs. Then I've downloaded i3 version, same here.

<details><summary>Fedora Linux 40 (Workstation Edition) (GNOME/gnome)</summary>

```ruby
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37100 (Git) AppImage
Build type: Release
Branch: main
Hash: ede6c811fd39d68a44949601f2eb7ecc8a67fdb7
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

</details>

<details><summary>Fedora Linux 40 (Forty) (i3/i3)</summary>

```ruby
OS: Fedora Linux 40 (Forty) (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37100 (Git) AppImage
Build type: Release
Branch: main
Hash: ede6c811fd39d68a44949601f2eb7ecc8a67fdb7
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

</details>

yomgui1, 2024-05-05:
Could be related to my window manager (i3). I'm going to make a debug build to run with gdb and try to understand.

yomgui1, 2024-05-05:
This is the full gdb stack:

```
#0  0x00007ffff6775f74 in boost::statechart::state_machine<Gui::GestureNavigationStyle::NaviMachine, Gui::GestureNavigationStyle::IdleState, std::allocator<boost::statechart::none>, boost::statechart::null_exception_translator>::terminate_impl (this=0x7ffff118a180 <QArrayData::shared_null>, theState=..., performFullExit=112) at /usr/include/boost/statechart/state_machine.hpp:935
#1  0x00007ffff6776606 in boost::statechart::state_machine<Gui::GestureNavigationStyle::NaviMachine, Gui::GestureNavigationStyle::IdleState, std::allocator<boost::statechart::none>, boost::statechart::null_exception_translator>::terminate_as_part_of_transit (this=0x7ffff118a180 <QArrayData::shared_null>, theState=...) at /usr/include/boost/statechart/state_machine.hpp:598
#2  0x00007ffff6773e5e in boost::statechart::simple_state<Gui::GestureNavigationStyle::AwaitingMoveState, Gui::GestureNavigationStyle::NaviMachine, boost::mpl::list<mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na>, (boost::statechart::history_mode)0>::transit_impl<Gui::GestureNavigationStyle::IdleState, Gui::GestureNavigationStyle::NaviMachine, boost::statechart::detail::no_transition_function> (this=0x4c19b90, transitionAction=...) at /usr/include/boost/statechart/simple_state.hpp:782
#3  0x00007ffff6772487 in boost::statechart::simple_state<Gui::GestureNavigationStyle::AwaitingMoveState, Gui::GestureNavigationStyle::NaviMachine, boost::mpl::list<mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na>, (boost::statechart::history_mode)0>::transit<Gui::GestureNavigationStyle::IdleState> (this=0x4c19b90)
    at /usr/include/boost/statechart/simple_state.hpp:313
#4  0x00007ffff676ffe3 in Gui::GestureNavigationStyle::AwaitingMoveState::react (this=0x4c19b90, ev=...) at /sources/freecad-src/src/Gui/GestureNavigationStyle.cpp:402
#5  0x00007ffff677f278 in boost::statechart::custom_reaction<Gui::GestureNavigationStyle::Event>::react<Gui::GestureNavigationStyle::AwaitingMoveState, boost::statechart::event_base, void const*> (stt=..., evt=..., 
    eventType=@0x7fffffffb1e8: 0x7ffff789b230 <boost::statechart::detail::id_holder<Gui::GestureNavigationStyle::Event>::idProvider_>) at /usr/include/boost/statechart/custom_reaction.hpp:41
#6  0x00007ffff677ea01 in boost::statechart::simple_state<Gui::GestureNavigationStyle::AwaitingMoveState, Gui::GestureNavigationStyle::NaviMachine, boost::mpl::list<mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na>, (boost::statechart::history_mode)0>::local_react_impl_non_empty::local_react_impl<boost::mpl::list<boost::statechart::custom_reaction<Gui::GestureNavigationStyle::Event>, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na>, boost::statechart::simple_state<Gui::GestureNavigationStyle::AwaitingMoveState, Gui::GestureNavigationStyle::NaviMachine, boost::mpl::list<mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na>, (boost::statechart::history_mode)0> > (stt=..., evt=..., 
    eventType=0x7ffff789b230 <boost::statechart::detail::id_holder<Gui::GestureNavigationStyle::Event>::idProvider_>) at /usr/include/boost/statechart/simple_state.hpp:814
#7  0x00007ffff677e05d in boost::statechart::simple_state<Gui::GestureNavigationStyle::AwaitingMoveState, Gui::GestureNavigationStyle::NaviMachine, boost::mpl::list<mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na>, (boost::statechart::history_mode)0>::local_react<boost::mpl::list<boost::statechart::custom_reaction<Gui::GestureNavigationStyle::Event>, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na> > (this=0x4c19b90, 
    evt=..., eventType=0x7ffff789b230 <boost::statechart::detail::id_holder<Gui::GestureNavigationStyle::Event>::idProvider_>) at /usr/include/boost/statechart/simple_state.hpp:850
#8  0x00007ffff677d347 in boost::statechart::simple_state<Gui::GestureNavigationStyle::AwaitingMoveState, Gui::GestureNavigationStyle::NaviMachine, boost::mpl::list<mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na, mpl_::na>, (boost::statechart::history_mode)0>::react_impl (this=0x4c19b90, evt=..., 
    eventType=0x7ffff789b230 <boost::statechart::detail::id_holder<Gui::GestureNavigationStyle::Event>::idProvider_>) at /usr/include/boost/statechart/simple_state.hpp:489
[... cut ...]
#40 0x00007ffff10bbacb in QCoreApplication::exec () at ../../include/QtCore/../../src/corelib/global/qflags.h:121
#41 0x00007ffff612f49d in (anonymous namespace)::tryRunEventLoop (mainApp=...) at /sources/freecad-src/src/Gui/Application.cpp:2027
#42 0x00007ffff612f752 in (anonymous namespace)::runEventLoop (mainApp=...) at /sources/freecad-src/src/Gui/Application.cpp:2047
#43 0x00007ffff612fa4f in Gui::Application::runApplication () at /sources/freecad-src/src/Gui/Application.cpp:2120
#44 0x000000000042653d in main (argc=1, argv=0x7fffffffd848) at /sources/freecad-src/src/Main/MainGui.cpp:278
```

yomgui1, 2024-05-06:
And that's happen only with gesture, I've tested other modes and it's fine.

Note also that if in GestureNavigationStyle.cpp, around line 400, I change the code for this:

```
auto r = transit<NS::IdleState>();
ns.openPopupMenu(ev.inventor_event->getPosition());
return r;
```

No more crash!

That's weird.

yomgui1, 2024-05-06:
It looks to be a bad memory overwriting, some internal values of "this" are completely random (like a bool with a value of 112).

maxwxyz, 2024-06-03:
@Rexbas maybe you can help related to gesture navigation?

Rexbas, 2024-06-08:
It looks like this problem only occurs with gesture navigation style and with transparent overlays active. I can reproduce with FreeCAD on my main screen but when I run FreeCAD on my secondary screen, there is no segfault when I click outside of FreeCAD.

I also found another way that causes the segfault. Right click in the 3D view to open the context menu. Then press left mouse button twice in the transparent part of the tree view.

I tried to understand why the segfault occurs but I did not succeed and therefore can't create a proper fix. However, it looks like `interceptEvent(..)` in the `OverlayManager` has something to do with it. 
https://github.com/FreeCAD/FreeCAD/blob/3b4598ce04d7aaec69feeb560c29ae18f0236e0c/src/Gui/OverlayManager.cpp#L1903

There are no more segfaults when commenting out `QApplication::sendEvent(..)` in this function.
https://github.com/FreeCAD/FreeCAD/blob/3b4598ce04d7aaec69feeb560c29ae18f0236e0c/src/Gui/OverlayManager.cpp#L1975

Unless someone, maybe @realthunder, has an idea what is causing the segfault. I think the best way forward is to implement the suggestion above by @yomgui1. I tried this and it prevents the segfault but I have no idea why.

xtemp09, 2024-06-08:
> I also found another way that causes the segfault. Right click in the 3D view to open the context menu. Then press left mouse button twice in the transparent part of the tree view.

@Rexbas, I cannot reproduce the crash you described.

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37665 (Git)
Build type: Debug
Branch: main
Hash: 62463fa0bd2c6962550ef67f94ca527b81b21cd9
Python 3.10.12, Qt 5.15.14, Coin 4.0.2, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

yomgui1, 2024-06-09:
I've just updated my Fedora from 38 to 40. I'm not able to build any more as the PySide2 support is dropped in this version, but I've tested the latest weekly build. It works well, not able to reproduce the crash with it on this F40.
It's a very good sign, at least I can continue to work without friction (I'm looking for a stable Qt6/PySide6 support ;-))
For the reason behind the crash, I suspect a memory corruption related to Qt5 version in F38.

Rexbas, 2024-06-09:
I also have Fedora 40 (KDE spin) on my laptop and I can still reproduce it on my laptop. And I tried various virtual machines where I can also perfectly reproduce it with the double click left mouse button on the transparent part of the tree view. Clicking outside FreeCAD only causes a segfault when I have two displays and then it only happens when FreeCAD runs on a particular display. It works fine on Windows 11.

xtemp09, 2024-07-13:
@Rexbas, could you repoduce the crash with [this commit](https://github.com/FreeCAD/FreeCAD/compare/main...xtemp09:FreeCAD:iss_13637)? Basicly, it's just a replacement of `exec()` with `popup()`.

Rexbas, 2024-07-13:
I cannot reproduce it with that commit, no more segfaults, nice work.

commit abb362f8 (2024-08-05): Fix the crash involving context menu

Closes #13637.
closed 2024-08-05 by adrianinsaval