FreeCAD/FreeCAD #13645: Sketcher: Can't join two arcs

dancesWithMachines, 2024-04-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Scenario:
1. Create one half of the arc and constrain it.
2. Add second arc using same point as center and join the "arms".

Expected results:
Arcs are joined together, sketch is constrained.

Actual result:
Cannot join the arcs.

Visual:
[image]

### Full version info

```shell
[code]
OS: macOS 14.4
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

Sketcher

### Anything else?

I believe this is a regression.
Two arcs cannot be joined in FreeCad 21.0+.
I verified this on 20.2 and it would be over-constrained but work.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-25:
can you check if you have enabled the setting auto remove constraints and play with it on/off?
Could you please also test this in the current dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

dancesWithMachines, 2024-04-25:
> can you check if you have enabled the setting auto remove constraints and play with it on/off?

I had `auto remove redundants` enabled by default. I disabled it and now it behaves like this.
On 21.2 I get over-constrain error like below.
[image]

On 20.02 I initially get same error, but if I try to move the sketch it resolves.
[image]

> Could you please also test this in the current dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

On [newest dev-build](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-36958-2024-04-23-conda-macOS-x86_64-py311.dmg) (36958-2024-04-23 )  it just crashes and exits.
[image]

Crash log from devbuild: [freecad_crash.zip](https://github.com/FreeCAD/FreeCAD/files/15113884/freecad_crash.zip)

maxwxyz, 2024-04-25:
@PaddleStroke FYI

dancesWithMachines, 2024-04-25:
In this particular scenario `equal` can be used to achieve same result, but this feels like workaround'ing. From logical point of view I don't see why joining two arcs would not work, it isn't intuitional. 
[image]
[image]

dancesWithMachines, 2024-04-25:
The scenario from this bug is very simple, but I originally come up from this.
[image]
Here the outcome is different, depending on what I do first.

PaddleStroke, 2024-04-26:
I totally agree that this feels very much like a bug. I have argued that in the past.
I'm just not sure how to solve it. Anyone know how other Cads are handling this situation? 

Maybe we need to add an exceptions in the solver for coincidence constraints for cases where arcs have 3 coincidences. Something like : 
```
if (arc has 3 coincidence)
      ignore one dof of center coincidence
```

Shkolik, 2025-02-28:
> Maybe we need to add an exceptions in the solver for coincidence constraints for cases where arcs have 3 coincidences. 

Sound like a good starting point. It's an edge case, but very frequently used one..

PaddleStroke, 2025-07-24:
@FlachyJoe any idea? What do you think of the suggestion I made above : 

> Maybe we need to add an exceptions in the solver for coincidence constraints for cases where arcs have 3 coincidences. Something like :
> 
> if (arc has 3 coincidence)
>       ignore one dof of center coincidence

@maxwxyz do you know how this behaves in other CADs?