FreeCAD/FreeCAD #13646: Sketcher: Offset problem

Pendelf, 2024-04-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Offset does not work correctly when selecting multiple contours (attached a link to a short video)

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git)
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.17
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * PieMenu 1.5.0
  * Pyramids-and-Polyhedrons
  * Reinforcement
  * sheetmetal 0.4.11
  * sheetmetal.backup1704622370.7006474 0.3.12
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://youtu.be/zxwOBCe8NbM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-25:
I guess it's currently working as intended, taking into account the largest outline. How should Sketcher know which geometries you want to include if you select all.
Workaround: Select each island and offset individually.
@PaddleStroke maybe there could be a checkbox/feature to include closed loops individually but that seems also confusing.

Pendelf, 2024-04-25:
Mostly there are problems when among the selected shapes there is a circle, but there are cases when everything works fine, for example two rectangles or a rectangle and a slot

Pendelf, 2024-04-25:
It also does not work well if the intersection mod is enabled

PaddleStroke, 2024-04-25:
From the video I see that the 2, 3 and 5th test are not behaving as
expected.
The 1st and 4th are doing what is expected.
I'm not sure why this happen. This is the result from occ so fixing that is
going to be hard.

PaddleStroke, 2024-04-26:
> It also does not work well if the intersection mod is enabled

Which is why its not the default option 😄 
I don't know why but indeed this mode does not get very reliable result from occ

bstott, 2025-11-07:
**Rev 1.1.0dev   Offset will not work for more than two segments of a multi-line polyline.**

I am having this problem. Offset is not working with multiple lines selected.  Offset will only select two lines out of a multi-segment polyline. Shown is three segment polyline. However, if it were a four segment polyline only two lines are still offset.

[image]

FreeCAD Configuration.

OS: Windows 10 build 19041
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/21 12:07:13
Build type: Release
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/100.707/1
Installed mods: 
  * 3D_Printing_Tools.backup1753665347.0583076
  * AddonManager 2025.11.4
  * AddonManager.backup1753665457.2166905 2025.6.1
  * Assembly4.backup1753996729.5129185 0.50.18
  * CAM
  * CfdOF.backup1753996894.4280593 1.30.4
  * CurvedShapes 1.0.14
  * Curves 0.6.70
  * Defeaturing 1.2.9
  * Design456 0.0.1
  * fasteners 0.5.43
  * fcVM 2024.9.5
  * freecad.gears 1.3.0
  * FusedFilamentDesign 0.25.380
  * InventorLoader 1.5.1
  * Launcher
  * Lithophane
  * MeshRemodel 1.10.38
  * OpticsWorkbench 1.2.2
  * PieMenu 1.12.1 (Disabled)
  * Quetzal 1.5.6
  * sheetmetal 0.7.58
  * Silk 0.2.0
  * SvgWorkbench 1.0.0.dev6
  * toSketch 1.0.1

PaddleStroke, 2025-11-12:
@bstott I think the problem you are mentioning is fixed already : https://github.com/FreeCAD/FreeCAD/pull/25091

closed 2025-11-12 by maxwxyz

Pendelf, 2025-11-13:
@maxwxyz 

It seems to me that the problem is still not solved, and bstott wrote about a multi-line with three or more segments, and now it really works

https://github.com/user-attachments/assets/32ef7d15-cd4c-45be-9684-46193130b622

bstott, 2025-11-13:
It works for what I'm doing now - I'm using version: 20251111.

[image]