FreeCAD/FreeCAD #13660: FEM: All the results pipeline objects are displayed simultaneously by default

FEA-eng, 2024-04-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

By default, all the results pipeline objects (multiple pipelines in transient, nonlinear and modal analyses or newly created filters under selected pipelines) have their visibility on, making it impossible to see anything meaningful.

I propose the following solution:
- only the first pipeline has visibility on for modal (frequency and buckling) analyses and only the last one has visibility on for transient and nonlinear analyses (since those results are primarily of interest) - in ParaView pipeline objects can have frames and they should be available in FreeCAD in the future too (#11055)
- only the last created filter under each pipeline has visibility on - that's how it works in ParaView as well

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git)
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-28:
also when e.g. creating more than one filters on the same result pipeline, on creation of the second one, the first result is completely displayed in red. Original color is restored when selecting the original filter in the tree.
demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/05d7cb55-b039-4a37-8810-6d12bac3afb0

FEA-eng, 2024-05-06:
I crossed out the second point as it was solved by #13820.

closed 2025-04-08 by FEA-eng