FreeCAD/FreeCAD #13661: TD Autofill, add the Current Sheet Number and the Total Sheet number as seperate option

tobiasfalk, 2024-04-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With #13005 Autofill was introduced, but it would be nice if the current Sheet number and the total number of sheet could be accessed separate from each other and not only as on, as it currently is.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git)
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * DynamicData 2.60.0
  * fasteners 0.5.15
  * kicadStepUpMod 11.1.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-05-06:
@tobiasfalk can you test https://github.com/FreeCAD/FreeCAD/commit/9bd19ff43a1c984d9d241f85c31c3d30be876d06 and report back ?

tobiasfalk, 2024-05-06:
Since can not Compile FreeCAD, I will need to wait until the next WeeklyBuild but if what I saw in the Pull Request #1366, it is now solved.

tobiasfalk, 2024-05-09:
Yes works, thx

closed 2024-05-09 by tobiasfalk

luzpaz, 2024-05-10:
Thanks @pavltom for the contribution!