FreeCAD/FreeCAD #13670: Tree view/Task view sidebars are unnecessary when there is no document

chennes, 2024-04-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Now that Start is no longer workbench, we load directly into the user's chosen workbench (Part Design by default). This means that that workbench's panels are all shown. In this case, when starting with no document, if the Start screen is shown it displays with the Task view and Tree view open, even though there is nothing in them. This is not a good use of space.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36992 (Git)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-04:
@chennes I think this is resolved?

chennes, 2025-05-04:
Start still shows with the various panels all open.