FreeCAD/FreeCAD #13676: Start: thumbnails for Recent and Examples are not loaded

furgo16, 2024-04-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In all the screenshots of the rewritten Start screen I've seen, the thumbnails for the files listed under Recent and Examples are shown. As they were in the previous Start workbench.

However, on my system the thumbnails are not shown. Either a gray background or a generic document icon is shown. I wonder if this is a Linux-specific issue. I can reproduce this both on my laptop and on my desktop.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36997 (Git) Snap 1000
Build type: Release
Branch: main
Hash: bff1efd665dc2f553ac57e5bef35c44a879999c2
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * ezydxf
  * Assembly4 0.50.12
  * ExplodedAssembly
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-04-27:
Please attach screenshot

chennes, 2024-04-28:
That seems like there is probably a permissions issue when the code tries to read the image from within the FCStd file. If your recent file is an image does it show that?

furgo16, 2024-04-28:
> If your recent file is an image does it show that?

1. I opened an image (.jpg) with File > Open...
2. The image was correctly imported on the XY plane. Its contents were correctly displayed
3. The image file did not appear in the Recent files section (that is, it did not appear on the list)
4. Closed FreeCAD, restarted it
5. The image file did not appear in the Recent files section (same as above)

Let me know if that answers the question.

furgo16, 2024-05-26:
@chennes did the above reply help? Is there any additional info I can provide to debug? Thanks!

80avin, 2024-06-06:
This also happens in snap installation. I also tried purging the data and reinstalling the snap, still the issue exists.
Screenshot
[image]

Full version info
```
OS: Ubuntu Core 22 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37655 (Git) Snap 1042
Build type: Release
Branch: main
Hash: de9be04249c862891d2a6903b5661460452796c7
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/India (en_IN)
Installed mods: 
  * CfdOF 1.25.17
  * Render 2024.1.23
  * sheetmetal 0.4.17
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * fasteners 0.5.21
  * 3D_Printing_Tools
  * Dracula 0.0.5
  * BillOfMaterials 0.0.8.3
  * woodworking 0.21.2.33771
  * Assembly4 0.50.12
  * OpenTheme 2024.5.16
  * PieMenu 1.6.0
```
(This happens without the installed addons also, and is happening since the Start page changed to new design)

furgo16, 2024-06-06:
Thanks for confirming. I was starting to wonder if it was my system only. I can reproduce it since the new start page was created, and with both .deb and .snap installs. If I build from source though, the issue is not reproducible.

80avin, 2024-06-06:
@chennes  I might be wrong but could it be because of these files missing ?
I was seeing open handles for the freecad process using `lsof` and found many handles saying `No such file or directory`

[image]

It'll be tedious for me to find each and every such file, but I can list all such files

Command:
```sh
lsof -p "$(pgrep FreeCAD)" | grep -E 'No such [^\)]+' | awk ' { print $8 }' | sort
```

Output
```
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.74.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libbreezecommon5.so.5.27.7
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libcanberra.so.0.2.5
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libcurl-gnutls.so.4.7.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libdbusmenu-qt5.so.2.6.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libdeflate.so.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libdouble-conversion.so.3.1
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libdrm_amdgpu.so.1.0.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libdrm_intel.so.1.0.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libdrm_nouveau.so.2.0.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libdrm_radeon.so.1.0.1
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libfam.so.0.0.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libfontconfig.so.1.12.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libglapi.so.0.0.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libGLdispatch.so.0.0.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libGL.so.1.7.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libGLX_mesa.so.0.0.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libGLX.so.0.0.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libgraphite2.so.3.2.1
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libharfbuzz.so.0.20704.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libICE.so.6.3.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libicudata.so.70.1
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libicui18n.so.70.1
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libicuuc.so.70.1
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libimobiledevice-1.0.so.6.0.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libjbig.so.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libjpeg.so.8.2.2
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5Archive.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5AuthCore.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5Bookmarks.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5Codecs.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5Completion.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5ConfigCore.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5ConfigGui.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5ConfigWidgets.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5CoreAddons.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5Crash.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5DBusAddons.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5GlobalAccel.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5GuiAddons.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5I18n.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5IconThemes.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5ItemViews.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5JobWidgets.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5KIOCore.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5KIOFileWidgets.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5KIOGui.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5KIOWidgets.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5Kirigami2.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5Notifications.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5Service.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5Solid.so.5.108.0
/snap/freecad/1042/kf5/usr/lib/x86_64-linux-gnu/libKF5Style.so.5.108.0
[... 107 lines cut ...]
/usr/lib/x86_64-linux-gnu/libgcrypt.so.20.3.4
/usr/lib/x86_64-linux-gnu/libgmp.so.10.4.1
/usr/lib/x86_64-linux-gnu/libgnutls.so.30.31.0
/usr/lib/x86_64-linux-gnu/libgpg-error.so.0.32.1
/usr/lib/x86_64-linux-gnu/libhogweed.so.6.4
/usr/lib/x86_64-linux-gnu/libidn2.so.0.3.7
/usr/lib/x86_64-linux-gnu/libkeyutils.so.1.9
/usr/lib/x86_64-linux-gnu/liblz4.so.1.9.3
/usr/lib/x86_64-linux-gnu/liblzma.so.5.2.5
/usr/lib/x86_64-linux-gnu/libmd.so.0.0.5
/usr/lib/x86_64-linux-gnu/libmpdec.so.2.5.1
/usr/lib/x86_64-linux-gnu/libnettle.so.8.4
/usr/lib/x86_64-linux-gnu/libp11-kit.so.0.3.0
/usr/lib/x86_64-linux-gnu/libpcre2-8.so.0.10.4
/usr/lib/x86_64-linux-gnu/libpng16.so.16.37.0
/usr/lib/x86_64-linux-gnu/libssh.so.4.8.7
/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30
/usr/lib/x86_64-linux-gnu/libsystemd.so.0.32.0
/usr/lib/x86_64-linux-gnu/libtasn1.so.6.6.2
/usr/lib/x86_64-linux-gnu/libtinfo.so.6.3
/usr/lib/x86_64-linux-gnu/libudev.so.1.7.2
/usr/lib/x86_64-linux-gnu/libunistring.so.2.2.0
/usr/lib/x86_64-linux-gnu/libz.so.1.2.11
/usr/lib/x86_64-linux-gnu/libzstd.so.1.4.8
```

80avin, 2024-06-06:
I just installed AppImage and it shows the thumbnails.
Also, `lsof` shows no file as `No such file or directory` (the above ones exist)
[image]

mfraser, 2024-07-29:
I'm not seeing thumbnails from the FreeCAD daily PPA either.

maxwxyz, 2024-11-03:
Is the issue also happening with the RC versions?

furgo16, 2024-11-03:
> Is the issue also happening with the RC versions?

Yes.

Shkolik, 2024-11-27:
as I described in #18153 thumbnails are missing on Win 11 with build 39315. All the settings/configs are the same

maxwxyz, 2024-12-03:
Started to appear on Windows now too in latest weekly:
[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.32
  * freecad.gears 1.2.0
  * Manipulator 1.5.7

```

furgo16, 2024-12-03:
Just to add some more data points:

**Snap package install**:
- Thumbnails of recent **image** files are now shown on the start page. I can no longer reproduce https://github.com/FreeCAD/FreeCAD/issues/15110, but I don't know where these thumbnails are stored either.
- Thumbnails of recent **.FCStd** files are still **not** shown on the start page
- There is a `$HOME/snap/freecad/common/.cache/thumbnails/normal` folder (probably as a result of https://github.com/FreeCAD/FreeCAD/pull/18095), but it's empty.
- There is also a `$HOME/snap/freecad/common/temp/FreeCADStartThumbnails` folder that has a few .png thumbnails. Suspiciously, the last thumbnail files in there are around the date I reported this issue. Could it be that something made FreeCAD stop writing on that location?

<details>

```OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Snap 1214
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```
</details>

**Deb package install**:
- Thumbnails of recent **image** files are now shown on the start page. I can no longer reproduce https://github.com/FreeCAD/FreeCAD/issues/15110, but I don't know where these thumbnails are stored either.
- There is a `$HOME/.cache/thumbnails/normal` folder (probably as a result of https://github.com/FreeCAD/FreeCAD/pull/18095), but it's empty. Note: I deleted the contents of `$HOME/.cache/thumbnails` before starting FreeCAD
- There is also a `$HOME/.cache/FreeCAD/Cache/FreeCADStartThumbnails` folder that has a few .png thumbnails. Suspiciously, the last thumbnail files in there are around the date I reported this issue. Could it be that something made FreeCAD stop writing on that location?

<details>

```
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39358 (Git)
Build type: Release
Branch: main
Hash: 735a07678d8682d5936906298ed61cb2efd0eb66
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * WikiPropertiesGenerator.FCMacro
  * fasteners 0.5.30
```

</details>

mfraser, 2024-12-03:
> Just to add some more data points:
> 
> **Snap package install**:
> 
>     * Thumbnails of recent **image** files are now shown on the start page. I can no longer reproduce [Start: recent images do not appear in Recent Files widget #15110](https://github.com/FreeCAD/FreeCAD/issues/15110), but I don't know where these thumbnails are stored either.
> 
>     * Thumbnails of recent **.FCStd** files are still **not** shown on the start page
> 
>     * There is a `$HOME/snap/freecad/common/.cache/thumbnails/normal` folder (probably as a result of [Start: Implement Thumbnail Managing Standard #18095](https://github.com/FreeCAD/FreeCAD/pull/18095)), but it's empty.
> 
>     * There is also a `$HOME/snap/freecad/common/temp/FreeCADStartThumbnails` folder that has a few .png thumbnails. Suspiciously, the last thumbnail files in there are around the date I reported this issue. Could it be that something made FreeCAD stop writing on that location?
> 
> 
> **Deb package install**:
> 
>     * Thumbnails of recent **image** files are now shown on the start page. I can no longer reproduce [Start: recent images do not appear in Recent Files widget #15110](https://github.com/FreeCAD/FreeCAD/issues/15110), but I don't know where these thumbnails are stored either.
> 
>     * Thumbnails of recent **.FCStd** files are still **not** shown on the start page
> 
>     * There is a `$HOME/.cache/thumbnails/normal` folder (probably as a result of [Start: Implement Thumbnail Managing Standard #18095](https://github.com/FreeCAD/FreeCAD/pull/18095)), but it's empty. Note: I deleted the contents of `$HOME/.cache/thumbnails` before starting FreeCAD
> 
>     * There is also a `$HOME/.cache/FreeCAD/Cache/FreeCADStartThumbnails` folder that has a few .png thumbnails. Suspiciously, the last thumbnail files in there are around the date I reported this issue. Could it be that something made FreeCAD stop writing on that location?

I have found that if I open the folder that contains the .FCStd files first so that the thumbnails are generated by Dolphin, FreeCAD will display the correct thumbnails.

Same goes for the example files, I had to open /usr/share/freecad-daily/examples/ so that the thumbnails were generated.

furgo16, 2024-12-03:
@mfraser could you use the About FreeCAD dialog to copy your full version information and paste it here? I assume you're using the .deb package, as that would not work with the .snap package. 

@80avin or @luzpaz could you perhaps edit https://github.com/FreeCAD/FreeCAD/issues/13676#issuecomment-2153365651 so that the file listing is collapsed using the [`<details>` markdown syntax](https://gist.github.com/scmx/eca72d44afee0113ceb0349dd54a84a2)? That comment makes the issue comments a bit difficult to read by making them unnecessarily long to scroll. Thanks!

mfraser, 2024-12-03:
> @mfraser could you use the About FreeCAD dialog to copy your full version information and paste it here? I assume you're using the .deb package, as that would not work with the .snap package.
> 
> @80avin or @luzpaz could you perhaps edit [#13676 (comment)](https://github.com/FreeCAD/FreeCAD/issues/13676#issuecomment-2153365651) so that the file listing is collapsed using the [`<details>` markdown syntax](https://gist.github.com/scmx/eca72d44afee0113ceb0349dd54a84a2)? That comment makes the issue comments a bit difficult to read by making them unnecessarily long to scroll. Thanks!

```
OS: Ubuntu 22.04.5 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39358 (Git)
Build type: Release
Branch: main
Hash: 735a07678d8682d5936906298ed61cb2efd0eb66
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * QuickMeasure 2022.10.28
  * kicadStepUpMod 11.4.3
  * Gridfinity 0.6.0
  * fasteners 0.5.33
  * dodo 1.0.1
  * Curves 0.6.51
  * ThreadProfile 1.92.0
  * sheetmetal 0.5.10
  * VendorParts
  * lattice2 1.0.0
  * Freecad-Built-in-themes-beta 1.2.2
  * Lithophane
  * ProDarkThemePreferencePack 1.0.0
  * Manipulator 1.5.7
  * DynamicData 2.70.0
  * FreeCAD-themes 2024.7.24
  * Defeaturing 1.2.2
  * Curves.backup1725432442.4530568 0.6.46 (Disabled)
  * CurvedShapes 1.0.13
  * Assembly4 0.12.4
  * freecad.gears 1.3.0
  * A2plus 0.4.68
  * fcgear 1.0.0
```

furgo16, 2024-12-05:
@maxwxyz @luzpaz could we please reopen this issue?

- The issue is not fixed in today's .snap builds, it's still reproduceable
- Daily PPA builds cannot be tested at the moment: https://github.com/FreeCAD/FreeCAD/issues/18308
- https://github.com/FreeCAD/FreeCAD/pull/18270, which automatically closed this issue upon being merged, was not aimed specifically at it, as far as I can tell. It was a follow-up fix [that was not supposed to fix this issue](https://github.com/FreeCAD/FreeCAD/pull/18095#issuecomment-2495854797).
- This was the original comment that might have contributed to conflate the two issues: https://github.com/FreeCAD/FreeCAD/issues/13676#issuecomment-2504143296. While the symptoms were the same, it was a different issue on a different platform.

furgo16, 2024-12-05:
On the snap package install the `$HOME/snap/freecad/common/.cache/thumbnails/normal` folder is now populated with .png files of various sizes, but they appear not to be valid .png (at least that's what the OS' image viewer complains about).

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39454 (Git) Snap 1216
Build type: Release
Branch: main
Hash: a97677ed2f0ee4875321f3043782cf77e3756cc6
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```

Shkolik, 2024-12-10:
Working fine now on:

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CurvedShapes 1.0.8
  * Curves 0.6.28
  * fasteners 0.5.21
  * Foamcut 0.1.5
  * Marz 0.1.5
  * OpenTheme 2024.9.1
  * PieMenu 1.8.1
  * SmartReportFields1.xlsx

[image]

maxwxyz, 2024-12-10:
@furgo16 can you confirm in latest weekly?

furgo16, 2024-12-10:
@maxwxyz still not working on Linux and either the .deb or .snap installation. That is, thumbnails are not shown on the start page. https://github.com/FreeCAD/FreeCAD/pull/18270 fixed it only for Windows (and perhaps other combinations of platform + packaging), but it did not address the original issue.

Reproduceable on the daily builds:

<details><summary>Snap daily build version</summary>

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39562 (Git) Snap 1221
Build type: Release
Branch: main
Hash: 16ac5f99770e078f97e07d7bcf06654f90846673
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```

</details>

<details><summary>Debian daily build version</summary>

```
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39358 (Git)
Build type: Release
Branch: main
Hash: 735a07678d8682d5936906298ed61cb2efd0eb66
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * WikiPropertiesGenerator.FCMacro
  * fasteners 0.5.30
```

</details>

In summary, 
- For .snap, `$HOME/snap/freecad/common/.cache/thumbnails/normal` is written with invalid .png (thumbnail) files
- For .deb, `$HOME/.cache/thumbnails/normal` is empty, so it would seem the .png (thumbnail) files are not written.

maxwxyz, 2024-12-10:
@wwmayer FYI

wwmayer, 2024-12-10:
@furgo16 which FC versions do you use?

furgo16, 2024-12-10:
@wwmayer each of my last comments has the full version info. I use daily builds from the main branch. See for instance the full about info for both the snap and .deb packages: https://github.com/FreeCAD/FreeCAD/issues/13676#issuecomment-2530827921

wwmayer, 2024-12-10:
Can you upload one of the invalid images from the snap version? The .deb package might be too old in order to include the latest fix.

furgo16, 2024-12-10:
@wwmayer thanks. Here are the 15 .pngs that are created on the `$HOME/snap/freecad/common/.cache/thumbnails/normal` folder when FreeCAD starts (I deleted the folder first). They are invariably 40 bytes or 8 bytes of size.

[Shome_snap_freecad_common__cache_thumbnails_normal.zip](https://github.com/user-attachments/files/18084934/Shome_snap_freecad_common__cache_thumbnails_normal.zip)

furgo16, 2024-12-10:
Update: unrelated, but it would appear that the freecad-daily PPA is stuck at revision 39358 for some reason. Perhaps an unwanted side effect of https://github.com/FreeCAD/FreeCAD-Bundle/issues/348? I just updated the .deb package to the latest version (which was higher than the one this morning), and it reports the same about info. It would seem that the package is being generated daily, but it builds the same git revision over and over. I'll investigate more and report an issue if that's the case.

If that's what's happening, this issue cannot be tested with the .deb package from the PPA for now.

maxwxyz, 2024-12-10:
@wwmayer @furgo16 it was reported here https://github.com/FreeCAD/FreeCAD/issues/18409 that it still does not work on Ubuntu24.04 with a newer build 39562.

furgo16, 2024-12-10:
Yes, [same build as I have](https://github.com/FreeCAD/FreeCAD/issues/13676#issuecomment-2530827921). It's the snap package from last night.

wwmayer, 2024-12-10:
I wonder whether this is a problem with the Snap packages. It has been confirmed that with AppImages and on Windows the thumbnails work now.

wwmayer, 2024-12-10:
> It is a problem with the .snap and .deb packages indeed

Well, I meant if the problem could be caused by the internal mechanism of the snap stuff?

furgo16, 2024-12-10:
I don't know unfortunately. It affects the .deb package as well, so it's not .snap-specific. Then again, it could be that the symptoms are the same, but they are separate issues for each package type.

I initially thought it could an issue that is Ubuntu-specific or Debian-based-distro-specific, but if I build the package locally I cannot reproduce the issue. So I wonder if it's caused by some sort of packaging policy.

What code does the writing of the .png thumbnails? Is it custom FreeCAD code, Qt, or something else?

furgo16, 2024-12-10:
Another data point: thumbnails had been working for both the .snap and .deb packages shortly up until the issue had been reported. It was around the time the new Start page was introduced, IIRC.

wwmayer, 2024-12-10:
>  It affects the .deb package as well

This only fails there because the available .deb files are too old.

> What code does the writing of the .png thumbnails? Is it custom FreeCAD code, Qt, or something else?

It starts here: https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Start/App/DisplayedFilesModel.cpp#L91

* getThumbnailsImage(): Returns the relative file name of the PNG image inside a project file
* getThumbnailsName(): Returns the relative path where to save the files. On Linux it's _thumbnails/normal_ as specified by freedesktop.org
* getThumnailsParentDir(): Returns the root directory where to create the _thumbnails/normal_
* getThumbnailsDir(): Returns the concatenated path of getThumnailsParentDir() and getThumbnailsName()
createThumbnailsDir(): Creates the above path in case it doesn't exist
* getMD5Hash(): Creates the MD5 hash of the FCStd file loaded by the user. This hash serves as filename of the PNG file as specified by freedesktop.org
* getUniquePNG(): Returns the absolute path of the PNG file
* useCachedPNG(): This checks if the PNG must be extracted from the FCStd file and saved or not
* loadFCStdThumbnail(): If needed extracts the PNG file, saves it to disk and returns its content to the calling instance 

> thumbnails had been working for both the .snap and .deb packages shortly up until the issue had been reported

Hmm, OK.

Gerd-St, 2024-12-11:
> Another data point: thumbnails had been working for both the .snap and .deb packages shortly up until the issue had been reported. It was around the time the new Start page was introduced, IIRC.

I can confirm this. Was not sure about as this was also the time where I went from Ubuntu 22.04 to 24.04 by a new installation.
And also on my system the thumbnail pngs at $HOME/snap/freecad/common/.cache/thumbnails/normal are not readable.

furgo16, 2024-12-11:
> This only fails there because the available .deb files are too old.

This comment was probably made before seeing https://github.com/FreeCAD/FreeCAD/issues/13676#issuecomment-2533209870 and https://github.com/FreeCAD/FreeCAD/issues/13676#issuecomment-2535248979. Even if the .deb package is too old (now reported at https://github.com/FreeCAD/FreeCAD/issues/18431), it does not change the fact that before reporting the issue thumbnails were working in the .deb and .snap packages.

To establish whether https://github.com/FreeCAD/FreeCAD/pull/18270 fixed the issue for the .deb package, we'll have to wait until the daily builds are fixed. Or until someone builds a local .deb with the current git HEAD to test it. I tried, but there seems to be quite a lot of [ git repository layers in Launchpad](https://code.launchpad.net/~freecad-maintainers/+git), and after spending a few time on it, I could not figure out where or how to do it.

> It starts here: https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Start/App/DisplayedFilesModel.cpp#L91
> [...]
> useCachedPNG(): This checks if the PNG must be extracted from the FCStd file and saved or not
> loadFCStdThumbnail(): If needed extracts the PNG file, saves it to disk and returns its content to the calling instance

Thanks! I'll start looking at these two methods, then. C++ is not my strength, so I can probably only diagnose.

80avin, 2024-12-11:
I have made a few observations, @wwmayer  can you or anyone comment on these ?
I can confirm it myself but it may take several weeks till I get the time to try building the snap myself.

### Observations:
* As I've pointed in https://github.com/FreeCAD/FreeCAD/issues/13676#issuecomment-2153365651 , I can see that the FreeCAD is not able to find some dynamic libraries provided by `kf5-5-108-qt-5-15-10-core22` plug.
   These include the libraries like `libjpeg.so.8.2.2`, `libpng16.so.16.37.0`, `libwebp.so.7.1.3` which I suspect will be required to draw images in QT.
* This, I suspect, is a misconfiguration in the [snapcraft.yaml](https://github.com/FreeCAD/FreeCAD-snap/blob/b8ddc79288d29a90354eba46b0f210e61103b7a4/snap/snapcraft.yaml) file.
   The doubt about misconfiguration is strengthened by few more observations:
   * FreeCAD is looking for `libpng16.so.16.37.0` in the directory `/usr/lib/x86_64-linux-gnu/`.
      However, I have `libpng16.so.16.43.0` in the same directory. On searching, I can find that this file exists in `/snap/core22/1722/usr/lib/x86_64-linux-gnu/libpng16.so.16.37.0` suggesting that snap is **misconfigured to look for `core22` dynamic libs in the system directory**
   * FreeCAD is looking for `libjpeg.so.8.2.2`, `libwebp.so.7.1.3` in `/snap/freecad/1222/kf5/usr/lib/x86_64-linux-gnu/` directory. However, this directory doesn't exist at all. I can confirm that these libs are present in `/snap/kf5-5-108-qt-5-15-10-core22/5/usr/lib/x86_64-linux-gnu/` directory. Suggesting that snap is **not mounting `/snap/kf5-5-108-qt-5-15-10-core22/5/` to `$SNAP/kf5`** or the path is wrong.
   Same observations can be made for other Not Found dependencies

The snap configuration has issues, this theory can further be strengthened by seeing the [snap build log](https://github.com/FreeCAD/FreeCAD-snap/actions/runs/12267426356/artifacts/2303415623). Unrelated to current issue, but seeing the error `ModuleNotFoundError: No module named 'pivy'` hints that something is wrong. Though, this may not be the cause of current issue

### My theory to why this might have happened ?
Because, probably, snap build is not linking to the plugs correctly.
Or, on ubuntu the dependencies are not being linked correctly.
Looking at source of [similar snapcrafts](https://github.com/search?q=path%3A**%2Fsnapcraft.y*ml+%22kf5%2Fusr%2Flib%2F%22+core22+&type=code), which use `core22` and `kf5-5-*` (  ), I can see 2 major differences with out [snapcraft.yaml](https://github.com/FreeCAD/FreeCAD-snap/blob/b8ddc79288d29a90354eba46b0f210e61103b7a4/snap/snapcraft.yaml)

* `SNAP_DESKTOP_RUNTIME: $SNAP/kf5` environment variable is not present in our snap.
* They use [kde-neon snap extension](https://snapcraft.io/docs/kde-neon-extension) to connect to `kf5-5-110-qt-5-15-11-core22` and use QT5

wwmayer, 2024-12-11:
> These include the libraries like libjpeg.so.8.2.2, libpng16.so.16.37.0, libwebp.so.7.1.3 which I suspect will be required to draw images in QT.

QtGui directly uses libpng16.so because PNG support is a fix part of Qt. But if the file was missing the application wouldn't even start. But independent on this issue it doesn't explain why the extracted PNG files are corrupted because this is independent from Qt stuff.

wwmayer, 2024-12-11:
> Shome_snap_freecad_common__cache_thumbnails_normal.zip

The content of the files show garbage similar like we had three weeks ago when the output file was opened without the `std::ios::binary` flag.

80avin, 2024-12-12:
@furgo16 I will do, only want to take time in refining them. This isn't my primary domain but I am finding snap to be very interesting.
Particularly, just learnt that I can build FreeCAD locally using snap in the cleanest(confined) way and also use vscode debugger with it.

While playing with vscode debugger only I noticed that I am not able to step into any of the zipios methods, even though the files appeared to be local.
This led to fixing the issue.

Thumbnails are shown when we set `-DFREECAD_USE_EXTERNAL_ZIPIOS=OFF`, which is default off but enabled only in
* snapcraft.yml: https://github.com/FreeCAD/FreeCAD-snap/blob/b8ddc79288d29a90354eba46b0f210e61103b7a4/snap/snapcraft.yaml#L165
* debian build: https://github.com/FreeCAD/FreeCAD/blob/d057fd79602acaa30a2b3e62cdac99b215cbd355/cMake/FreeCAD_Helpers/InitializeFreeCADBuildOptions.cmake#L180-L184

Video
https://github.com/user-attachments/assets/235bdd60-3adf-4322-a8ba-4325204a3287

Before the Start WB migration, we were using python zipFile library that's why didn't encounter this.
Does this work as a solution, or is there still another hidden bug in FreeCAD which might've caused this ?

furgo16, 2024-12-12:
Wow, great detective work @80avin!

@wwmayer @chennes what are your thougths on the above?

chennes, 2024-12-12:
I have no idea why the Snap is being created with that flag set --  if you submit a PR to Snap repo that turns it off I'll merge it and we can see if anything explodes.

furgo16, 2024-12-12:
I'm testing this change on a separate branch at https://github.com/FreeCAD/FreeCAD-snap/compare/master...furgo16:FreeCAD-snap:refs/heads/zipios-off

I've triggered a build on that branch. If all goes well, it should be ready in ~1:30. Once finished, a snap for local testing should be available under the "Artifacts" section on the workflow run summary page at https://github.com/furgo16/FreeCAD-snap/actions/runs/12306245529

furgo16, 2024-12-13:
## Deb package

I've upgraded the .deb package on my system after the https://github.com/FreeCAD/FreeCAD/issues/18455 fix. It now ships up-to-date code that includes the https://github.com/FreeCAD/FreeCAD/pull/18270 fix:

```
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39585 (Git)
Build type: Release
Branch: main
Hash: d057fd79602acaa30a2b3e62cdac99b215cbd355
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * WikiPropertiesGenerator.FCMacro
  * fasteners 0.5.30
```

The issue is still reproduceable, so I think we can establish that https://github.com/FreeCAD/FreeCAD/pull/18270 did not fix the issue in the .deb package.

Next: try if removing this snippet from cmake fixes it, as per @80avin's comment

```
     if (FREECAD_BUILD_DEBIAN) 
         set(FREECAD_USE_EXTERNAL_ZIPIOS ON ) 
         # A Debian package for SMESH doesn't exist 
         #set(FREECAD_USE_EXTERNAL_SMESH ON ) 
     endif (FREECAD_BUILD_DEBIAN) 
```

## Snap package

The local testing snap with `FREECAD_USE_EXTERNAL_ZIPIOS OFF` is now available under the "Artifacts" section on the workflow run summary page at

https://github.com/furgo16/FreeCAD-snap/actions/runs/12306245529

I'll test it during the day.

furgo16, 2024-12-13:
Good point, as the snap package also bundles that package. But looking at the package version numbers at https://packages.debian.org/search?searchon=sourcenames&keywords=zipios%2B%2B, the package has not had any changes in the last 4 Debian releases.

I can submit the PR to remove the external zipios from the snap to address it, but still don't understand why using the external zipios stopped working at some point. @80avin you mentioned something that might be the cause:

> Before the Start WB migration, we were using python zipFile library that's why didn't encounter this.

Could you expand on this?

wwmayer, 2024-12-13:
In the meantime I have built the external zipios package and I can confirm that it fails to extract the PNG file from an FCStd file.

The function to blame is `ZipFile::getInputStream`. In the Debian sources its implementation is:
```
istream *ZipFile::getInputStream( const string &entry_name, 
				  MatchPath matchpath ) {
  if ( ! _valid )
    throw InvalidStateException( "Attempt to use an invalid ZipFile" ) ;

  ConstEntryPointer ent = getEntry( entry_name, matchpath ) ;
  
  if ( ent == 0 )
    return 0 ;
  else {
    ZipInputStream *zis( new ZipInputStream( _filename,
      static_cast< const ZipCDirEntry * >( ent.get() )->
      getLocalHeaderOffset() + _vs.startOffset() ) ) ;
    zis->getNextEntry();
    return zis;
  }
}
```
while in the FreeCAD sources it's:
```
istream *ZipFile::getInputStream( const string &entry_name, 
				  MatchPath matchpath ) {
  if ( ! _valid )
    throw InvalidStateException( "Attempt to use an invalid ZipFile" ) ;

  ConstEntryPointer ent = getEntry( entry_name, matchpath ) ;
  
  if ( !ent )
    return nullptr ;
  else
    return new ZipInputStream( _filename,	
			   static_cast< const ZipCDirEntry * >( ent.get() )->
			   getLocalHeaderOffset() + _vs.startOffset() ) ;
}
```
The actual difference between them is that the call of `zis->getNextEntry();` has been dropped. If I comment out this line in the Debian source and rebuild the package it fixes the problem.

> the package has not had any changes in the last 3 Debian releases.

But it looks like a change that was made by the Debian maintainers many years ago that has never been ported to the FreeCAD sources. If I look at the time when zipios++ has been added to FreeCAD then besides some code clean-up the `ZipFile::getInputStream` hasn't been changed: https://github.com/berndhahnebach/1_gitcsv/commit/57970d2092702dce8242a2e5e0373040d3c74789

So, the question is why and when the Debian version has been changed.

wwmayer, 2024-12-13:
When looking at [upstream](https://github.com/Zipios/Zipios/blob/15aec038e631952c822a7991707afad8a4d3adc3/src/zipfile.cpp#L560) then not even there is this additional call of getNextEntry().

But when looking at the history of the file it has been changed with https://github.com/Zipios/Zipios/commit/c80a6d0c5218d7f1a5366fb39bd637166457814b in 2008 and reverted again with the next change https://github.com/Zipios/Zipios/commit/b4c66a21adff7b965947f6b4a7181fe9c71c915a in 2015.

So, in other words the Debian maintainers have picked up the broken version and never reverted that change.

furgo16, 2024-12-13:
> So, the question is why and when the Debian version has been changed.

If I'm reading the [Debian changelog](https://metadata.ftp-master.debian.org/changelogs//main/z/zipios++/zipios++_0.1.5.9+cvs.2007.04.28-10+deb10u1_changelog) correctly:

```
zipios++ (0.1.5.9+cvs.2004.02.07-1) unstable; urgency=low

  * New upstream release (CVS snapshot).
  * Bumped Standards-Version to 3.6.1.

 -- Masayuki Hatta (mhatta) <mhatta@debian.org>  Sat,  7 Feb 2004 16:47:39 +0900

zipios++ (0.1.5+cvs.2003.03.18-1) unstable; urgency=low

  * New upstream release (CVS snapshot).
  * Added Build-Depends: automake1.7 - closes: #178638

 -- Masayuki Hatta <mhatta@debian.org>  Tue, 18 Mar 2003 10:51:25 +0900
```

This change occured in 2004. If I'm not mistaken, what I think what might add to the confusion is that:

- FreeCAD bundles the 0.1.5 version. That corresponds mainly to the upstream source plus some minor changes and a CVE patch.
- Debian ships the 0.1.5.9 snapshot sources. There was no official 0.1.5.9 release as far as I can tell. But the zipfile.cpp file in the Debian package corresponds to that in the 0.1.7 upstream source.

So that would explain why they are different. But it doesn't explain why upstream made the change in 0.1.5.9 

The zipios++ project is now [hosted in GitHub](https://github.com/Zipios/Zipios) and is at version 2.2.x, but the old 0.1.x versions can still be downloaded from the old location at Sourceforge: https://sourceforge.net/projects/zipios/files/zipios%2B%2B/. Reportedly, the git repo on Github has got the full history, but the old releases seems not to have been tagged or branched, so it would require some detective work to find them in git.

furgo16, 2024-12-13:
I might have been wrong, but it turned out to be relatively easy to search the upstream sources.

I think the related changes were made in 2007 and correspond to these 3 commits:
- https://github.com/Zipios/Zipios/commit/ae6f08d
- https://github.com/Zipios/Zipios/commit/4751acb
- https://github.com/Zipios/Zipios/commit/c80a6d0

furgo16, 2024-12-13:
Ah, sorry, I had not seen https://github.com/FreeCAD/FreeCAD/issues/13676#issuecomment-2541325750 before I wrote my comments

wwmayer, 2024-12-13:
Basically what has happened is this:
* Before all the trouble the constructor of ZipInputStreambuf had a call of getNextEntry() while ZipFile::getInputStream() simply created the ZipInputStream.
* For some unknown reason the call of getNextEntry() has been removed from ZipInputStreambuf
* This must have caused some problems so that ZipFile::getInputStream() was changed to call getNextEntry()
* Debian fetched this state of the upstream version ([Import Upstream version 0.1.5.9+cvs.2007.04.28-1](https://salsa.debian.org/debian/zipios/-/commit/62754b6ba016b98ed06822440a81a0e485237961?page=3)) 
* Another Debian package using libzipios++ was broken so that the change in ZipInputStreambuf has been reverted ([Import Debian changes 0.1.5.9+cvs.2007.04.28-2](https://salsa.debian.org/debian/zipios/-/commit/930a63bdf22d79bf501479a7227e0cc2d2b793f3))
* However, the change in ZipFile::getInputStream() has never been reverted so that we end up with two calls of getNextEntry().

wwmayer, 2024-12-13:
I have reported it to the Debian maintainer right now. I will post the link once it appears in their bug tracking system...

furgo16, 2024-12-13:
Great, thanks!

80avin, 2024-12-13:
@furgo16 
Not sure if it is still required but since you asked,
> > Before the Start WB migration, we were using python zipFile library that's why didn't encounter this.
>
> Could you expand on this? Did this masked the fact that using the external zipios++ in conjunction with reading/writing thumbnails perhaps never worked in the first place?

https://github.com/FreeCAD/FreeCAD/blob/1f9be3f9ed44f7a3f4a453ef4761eea2d96a3dc0/src/Mod/Start/StartPage/StartPage.py#L188
This is the older code which was removed in https://github.com/FreeCAD/FreeCAD/pull/13134

> Did this masked the fact that using the external zipios++ in conjunction with reading/writing thumbnails perhaps never worked in the first place?

Already deduced. Not just thumbnails, but nothing could be extracted with external ZIPIOS. I tried this with [getMetadata](https://github.com/FreeCAD/FreeCAD/blob/4e82a0af4863363dd9e31e9660fb66ae7383bc23/src/App/ProjectFile.cpp#L245-L254) which also worked only after I used internal zipios.

furgo16, 2024-12-14:
The fix is already available in the last snap daily package, and it works well. However, a manual step is required to remove the previously created corrupted thumbnail images:

```shell
rm -Rf ~/snap/freecad/common/.cache/thumbnails/normal
```

Once deleted, FreeCAD will create the correct thumbnails upon start. I wonder if there is a way to print a warning that the .png files could not be correctly be read. That would have saved a lot of debugging time on this issue.

Now only the .deb package is left to do to close the issue. I'd suggest simply disabling the external zipios for a minimal change. In this case, if and when the Debian package is fixed, the switch can simply be turned back on in the future.

```diff
     if (FREECAD_BUILD_DEBIAN) 
-         set(FREECAD_USE_EXTERNAL_ZIPIOS ON ) 
+         set(FREECAD_USE_EXTERNAL_ZIPIOS OFF ) 
         # A Debian package for SMESH doesn't exist 
         #set(FREECAD_USE_EXTERNAL_SMESH ON ) 
     endif (FREECAD_BUILD_DEBIAN) 
```

closed 2024-12-14 by chennes

adrianinsaval, 2024-12-14:
I assume this is still a problem for the ppa?

furgo16, 2024-12-15:
@adrianinsaval the issue in the .deb packages seems to have been fixed with https://github.com/FreeCAD/FreeCAD/pull/18494 after today's freecad-daily build

After upgrading the packages, I deleted `$HOME/.cache/thumbnails/normal/`, started `freecad-daily` and I could finally see the thumbnails again.

```
$ apt show freecad-daily

Package: freecad-daily
Version: 1.1~pre1~202412150103~ubuntu22.04.1
Priority: optional
Section: science
Maintainer: Debian Science Maintainers <debian-science-maintainers@lists.alioth.debian.org>
Installed-Size: 33,8 kB
Depends: freecad-daily-python3
Recommends: calculix-ccx, graphviz, git
Breaks: freecad-daily (<< 0.22)
Replaces: freecad-daily (<< 0.22)
Download-Size: 15,5 kB
APT-Manual-Installed: yes
APT-Sources: https://ppa.launchpadcontent.net/freecad-maintainers/freecad-daily/ubuntu jammy/main amd64 Packages
[...]
```

That said, there was a specific file where the thumbnail was not saved. I need to investigate that one, and I'll report back if it's an actual issue.

All in all, I think we can finally declare this issue closed :tada:

wwmayer, 2024-12-21:
The Debian maintainer has applied the fix today.