FreeCAD/FreeCAD #13677: TechDraw: Axonometric length dimension problem

Pendelf, 2024-04-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After deleting the axonometric length dimension, the anchor points remain
Attached a link to the video 
[image]
[image]

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36996 (Git)
Build type: Release
Branch: main
Hash: d92f74ef019fcedbdcdc32055ab2437c950ac23b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.17
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * PieMenu 1.5.0
  * Pyramids-and-Polyhedrons
  * Reinforcement
  * sheetmetal 0.4.11
  * sheetmetal.backup1704622370.7006474 0.3.12
```

### Subproject(s) affected?

Techdraw

### Anything else?

https://youtu.be/Z2xpWpw3dAk

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-04-27:
Do you mind zipping up and drag/dropping an example file into this ticket ?

Pendelf, 2024-04-28:
Okay, I've added the archive

maxwxyz, 2024-04-28:
The pages are empty in both files for me with 

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git)
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.35
  * dodo 1.0.1
  * fasteners 0.5.20
  * fcVM-WB
  * freecad.gears 1.2.0
  * OpenTheme 2024.4.20
  * sheetmetal 0.4.13

``` 

so I cannot reproduce. Also after adding a template the views are blank, not sure if the file is bad or recent changes in TechDraw might cause this behavior on my side @WandererFan FYI

Pendelf, 2024-04-29:
Tried it on linux laptops, same result

OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36996 (Git) AppImage
Build type: Release
Branch: main
Hash: d92f74ef019fcedbdcdc32055ab2437c950ac23b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * fasteners 0.4.75
  * ExplodedAssembly
  * BIM 2021.12.0
  * CurvedShapes 1.0.5
  * sheetmetal 0.3.13
  * dodo 1.0.0
  * Curves 0.6.21
  * A2plus 0.4.61
  * LCInterlocking
  * lattice2 1.0.0

[image]
[image]

luzpaz, 2024-08-25:
Hi @pavltom would you have anytime to dig into this ? TIA!

FEA-eng, 2025-09-29:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

luzpaz, 2025-11-17:
> quick analysis:

@WandererFan is this a simple fix ?

Pendelf, 2026-01-06:
Seems like it's still relevant

https://github.com/user-attachments/assets/7f02eb5c-f054-48ad-b97c-89a546b6a4f4

OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251230 (Git shallow)
Build date: 2025/12/30 17:40:39
Build type: Release
Branch: grafted,grafted
Hash: f5759e580b6330906beca12b4be419b030a088ab
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * addFC 1.9.3
  * CurvedShapes 1.0.14
  * Curves 0.6.70
  * fasteners 0.5.43
  * PieMenu 1.12.1
  * sheetmetal 0.7.58
  * StructureTools 0.0.1
  * SvgWorkbench 1.0.0.dev6
  * Telemetry 1.0.4