FreeCAD/FreeCAD #13680: FreeCAD does not build with shiboken6  6.7

mark-herbert42, 2024-04-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

/var/tmp/portage/media-gfx/freecad-0.21.99/work/FreeCAD-0.21.99/src/Gui/PythonWrapper.cpp: In function ‘bool Gui::loadPySideModule(const std::string&, PyTypeObject**&)’:
/var/tmp/portage/media-gfx/freecad-0.21.99/work/FreeCAD-0.21.99/src/Gui/PythonWrapper.cpp:305:43: error: cannot convert ‘Shiboken::Module::TypeInitStruct*’ to ‘PyTypeObject**’ {aka ‘_typeobject**’} in assignment
  305 |         types = Shiboken::Module::getTypes(requiredModule);
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
      |                                           |
      |                                           Shiboken::Module::TypeInitStruct*

### Full version info

```shell
builing on gentoo, qt6.7 , shiboken/pyside 6/7

It was building OK with qt 6.6.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-04-27:
Related: 

* https://github.com/FreeCAD/FreeCAD/issues/13303
* https://github.com/FreeCAD/FreeCAD/issues/13427
* https://github.com/FreeCAD/FreeCAD/issues/6992

wwmayer, 2024-04-28:
https://github.com/FreeCAD/FreeCAD/pull/13647

closed 2024-04-30 by adrianinsaval