FreeCAD/FreeCAD #13681: Regression: Marker size and Pick Radius in different places in the preferences

chrisb-github, 2024-04-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Marker size and Pick radius are so closely related that it is not sensible to have them on different pages.

History: In the old days (0.17 or so) these two settings were on different pages. This was always a hassle to explain, because they almost always are changed together.

In 0.20 they were very sensible placed directly above each other:
[image]

In 0.21 the were divided again and this is still the case in 0.22.

I see that for formal reasons this could be divided, but it is not user friendly at all.

Possible solutions:

1. (Preferred) Place Pick Radius _and_ Marker size on both pages, General->Selection _and_ Display->3D view. They are both not too crowded.
2. Place them both on Display->3D view
3. Place them both on General->Selection
4. Make them both a single setting which can be finetuned in Tools->Parameters.

### Full version info

```shell
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36904 (Git)
Build type: Release
Branch: main
Hash: 0622cb784e199053656b50854e92eff806130408
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.2.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fasteners 0.5.15
  * ExplodedAssembly
  * Curves 0.6.29
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-04-28:
To me the change also made very little sense. The pick radius is not only used for selection but for any kind of picking into the 3D view, e.g. when defining the points for measurements. So, in this context the move is quite illogical.

For more details:
https://github.com/FreeCAD/FreeCAD/issues/9205
https://github.com/FreeCAD/FreeCAD/pull/9237

wwmayer, 2024-04-28:
Btw, this is the PR that once fixed it and its associated ticket:
https://github.com/FreeCAD/FreeCAD/pull/3932
https://tracker.freecad.org/view.php?id=4147

nate-i-think, 2025-12-26:
This should absolutely be changed. As a new user I just spend 10 minutes confused about this.

maxwxyz, 2026-01-18:
If someone wants to cherry-pick https://codeberg.org/xCAD/FreeCAD11/pulls/129