FreeCAD/FreeCAD #13685: Version 0.21.2 does not run save with Windows 8.1 operating system

rofreecad, 2024-04-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I installed the 64-bit installer version 0.21.2 encountering an inability to save the job.
Running the Save command keeps the program waiting for a response that does not come.
This leaves the only option to terminate the program execution. 

Tried with the 64-bit portable version 0.21.2 everything works normally.
Tried uninstalling everything, program and settings, but the problem does not resolve.
From the tests performed everything works up to version 0.20.0, all subsequent versions do not perform data saving.

### Full version info

```shell
[code]
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Local: Italian/Italy (it_IT)
Installed mods: 
  * fasteners 0.5.17
  * Help 1.0.3
  * kicadStepUpMod 11.1.2
  * sheetmetal 0.4.11
[/code]

[code]
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Local: Italian/Italy (it_IT)
Installed mods: 
  * fasteners 0.5.17
  * Help 1.0.3
  * kicadStepUpMod 11.1.2
  * sheetmetal 0.4.11
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-04-30:
This sounds like a Windows permissions issue. It may also be due to the folder that was last used for saving.

rofreecad, 2024-05-01:
I thank you for the feedback.
In fact I am verifying that on the other disk partitions it saves regularly.
Only on the operating system's C: partition it doesn't because of insufficient permissions.
On the compiuter I am the only user and also administrator so I should be able to do it.
It also seems to me a problem introduced by the compilation of the installer since until version 0.20.0 there was not this behavior.

luzpaz, 2024-07-22:
Keep open ?

rofreecad, 2024-07-22:
I checked again, confirm that it does not allow saving to the Windows C: partition.
It normally saves the file in the other partitions.
[code]

OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * Assembly4 0.50.13
  * fasteners 0.5.23
  * Help 1.0.3
  * kicadStepUpMod 11.1.2
  * sheetmetal 0.4.11
[/code]

Roy-043, 2024-08-25:
This is a Windows permissions issue. It has nothing to do with FreeCAD.

closed 2024-08-25 by Roy-043