FreeCAD/FreeCAD #13690: Reordering top level objects destroys the Placement

wwmayer, 2024-04-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

See https://forum.freecad.org/viewtopic.php?t=87165
1. Go to Sketcher wb
2. Create a new sketch and set its offset to 10. Leave edit mode.
3. Create a second sketch. Leave edit mode.
4. Copy & paste the first sketch
5. Drag & drop the copied sketch on the second sketch so that it's moved at the end
6. Placement is lost

The forum thread mentions two other issues with the tree ordering but this is the most serious issue.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36999 (Git)
Build type: debug
Branch: issue_13107
Hash: ed77603af999525b65dcfc4a41337f15a052c9ef
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-04-28:
This is the line where the placement is reset:
https://github.com/FreeCAD/FreeCAD/blob/ed77603af999525b65dcfc4a41337f15a052c9ef/src/Gui/Tree.cpp#L2154C25-L2154C80

What happens there is that the identity matrix is set and not the original matrix. The reason for this is here:
https://github.com/FreeCAD/FreeCAD/blob/ed77603af999525b65dcfc4a41337f15a052c9ef/src/Gui/Tree.cpp#L2058

Why on earth does it depend on user settings that for a simple object the placement is kept or reset? This is awful behaviour.

maxwxyz, 2024-04-28:
@PaddleStroke FYI

PaddleStroke, 2024-04-30:
I don't know. I only moved this code when refactoring the `dropEvent` function. See the file before my PR : 

https://github.com/FreeCAD/FreeCAD/blob/560898907b317893d90e3a031c41ea1d63109dd9/src/Gui/Tree.cpp#L2321

https://github.com/FreeCAD/FreeCAD/blob/560898907b317893d90e3a031c41ea1d63109dd9/src/Gui/Tree.cpp#L2411

wwmayer, 2024-04-30:
> I only moved this code when refactoring the dropEvent function. See the file before my PR

I know it's there for almost 5 years. Until now it never caused problems (that I am aware of). But now it does and it must be checked why. The specific points to check with a version prior to your change when sync placement is off:
* Does moving a top-level object to a container object (part, body, group, ...) always keep the placement?
* Does moving a child object to top-level or another container always keep the placement?
* Is the intention of this option to reset the placement only when moving to another document?
* For all code paths of  the old dropEvent() a use case must be found to better understand the behaviour and intention.

If we better understand the details the behaviour must be compared with the new code.

wwmayer, 2024-05-05:
With the old drag and drop implementation it was explicitly checked that if a top-level item is moved to its parent document item then it was filtered out.
```
            auto parentItem = item->getParentItem();
            if (!parentItem) {
                if (da == Qt::MoveAction && obj->getDocument() == thisDoc)
                    continue;
            }
```

Now, changing the tree order of a top-level item only works by moving it to its parent document item and therefore won't be filtered out any more. Now, as side-effect the code to reset the placement will be executed.

PR #13835 (2024-05-06): Gui: Do not reset placement when reordering top-level objects
This fixes #13690: Reordering top level objects destroys the Placement
commit 638cbd02 (2024-05-06): Gui: Do not reset placement when reordering top-level objects

This fixes #13690: Reordering top level objects destroys the Placement
closed 2024-05-06 by yorikvanhavre