FreeCAD/FreeCAD #13694: FEM: Data along line - orbit cancels selection of points

maxwxyz, 2024-04-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the DataAlongLine filter in a results pipeline it is possible to select the two points in the 3d view.
If you orbit the camera, the operation is cancelled, either before you've placed the first point or also after you've placed the first point.
You should be able to navigate the model when selecting the points.

@FEA-eng FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git)
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.35
  * dodo 1.0.1
  * fasteners 0.5.20
  * fcVM-WB
  * freecad.gears 1.2.0
  * OpenTheme 2024.4.20
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-05-27:
I can confirm this, it indeed can be quite annoying.

@marioalexis84 FYI

marioalexis84, 2024-06-04:
@maxwxyz @FEA-eng Which navigation style are you using?

maxwxyz, 2024-06-05:
CAD, orbit around window center.

FEA-eng, 2024-06-05:
Panning with CAD style also causes it.

PR #15767 (2024-08-05): Fem: Remove callback on right click during point marker selection - fixes #13694
Fixes #13694.
Point is selected with left click and cancelled with right click for both Data At point clip filter and Line clip filter.

@maxwxyz @FEA-eng
commit ca70e06a (2024-08-05): Fem: Remove callback on right click during point marker selection - fixes #13694
closed 2024-08-05 by chennes