FreeCAD/FreeCAD #13696: Draft: Cursor / Pointer Icon huge on HiDPI Screens

maxwxyz, 2024-04-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the Draft WB tools to create geometry, the cursor icon / mouse pointer is displayed very large on HiDPI screens.
It looks normal on my normal DPI screen. Sketcher pointer icons look the same on both screens, as they seem to respect the DPR.
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36958 (Git)
Build type: Release
Branch: main
Hash: a041129090b8dcbb8367636840d3c668ef30a933
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.35
  * dodo 1.0.1
  * fasteners 0.5.20
  * fcVM-WB
  * freecad.gears 1.2.0
  * OpenTheme 2024.4.20
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-04-30:
We have tried to fix an issue with the Draft cursor on the Mac (#10962). That fix may have introduced a regression. Can you test this with V0.21?

maxwxyz, 2024-05-01:
@Roy-043 can confirm, the cursor is not huge in 0.21.2.
But the cursor is tiny there, on my HiDPI screen, so also not good.
[image]

Sketcher cursor icons look correctly scaled in 0.21.2 and 0.22dev, they use the DPR.

<details><summary>0.21.2 version info</summary>
<p>

```
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.35
  * dodo 1.0.1
  * fasteners 0.5.20
  * fcVM-WB
  * freecad.gears 1.2.0
  * OpenTheme 2024.4.20
  * sheetmetal 0.4.13
[/code]

```

</p>
</details>

Roy-043, 2024-07-31:
I think there may be something wrong with the DPR (pixel ratio) calculation in Draft.

Judging from the images you have posted [here](https://github.com/FreeCAD/FreeCAD/issues/11352), your DPR for the Sketcher cursor is 2.

I think that for the Draft cursor your DPR may be 4. Can you verify please?:
1. Switch to the Draft WB.
2. Create a Draft Line.
3. Paste `Gui.Snapper.device_pixel_ratio()` in the Python console.

maxwxyz, 2024-08-01:
@Roy-043 
```
>>> # Gui.runCommand('Draft_Line',0)
>>> Gui.Snapper.device_pixel_ratio()
2
```

closed 2024-08-08 by Roy-043