FreeCAD/FreeCAD #13699: bspline coming out as straight line in print

theosib, 2024-04-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have used a bspline to join two arcs. There are two segments to the bspline, and each segment has been made tangent to its adjoining arc. The spline looks just fine in the sketcher, but in both the part design view and the exported STEP file, it comes out as a straight line.

I've attached three images. The one from the sketcher looks fine. But other two are from the part design view and from the slicer, and both have a weird kink.

[image]
[image]
[image]

BTW, I would be happy to upload the FCStd file here, but I get a "we don't support that file type" error.

### Full version info

```shell
[code]
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-04-28:
Here's the cad file zipped.
[card case.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/15143904/card.case.FCStd.zip)

maxwxyz, 2024-04-29:
@AjinkyaDahale FYI

AjinkyaDahale, 2024-05-09:
Is this fixed if we use, say, a similarly shaped polyline instead?

AjinkyaDahale, 2024-05-10:
The biggest issue is the discretiziation. The splines are extremely skewed, and thus have a lot of variability in curvature. However, the discretization is (by my guess) done based on the parameter, which is not fine enough where it's needed.

Here are some possibilities:
1. Increase the degree of each spline, and try to adjust after that, so that the curvature doesn't change that drastically. A cubic spline should do very well. There will still be some degrees of freedom left after all current are applied, but maybe try block constraint after.
2. Try changing the curvature or position of the smaller arcs.
[image]

AjinkyaDahale, 2025-07-20:
"Not planned" is probably appropriate. Also likely: OCCT or Coin limitation. @pieterhijma probably a draw harness could be useful to see if this also happens on OCCT.

AjinkyaDahale, 2025-07-21:
I'll defer this to Pieter. Likely this is already the issue for that.