FreeCAD/FreeCAD #13700: Draft WB UI: inconsistent temporary Object creation in the Tree View

marcuspollio, 2024-04-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating Drafting Objects (via Toolbar, Shortcuts or Menu), the Line, Polyline, B-Spline and Bézier curve Tools create a temporary "Part_3D_object" icon in the Tree View, while the other Tools (such as Arc, Circle, Rectangle, etc.) do not.
When the Task is finished (e.g. your Line is fully drawn), the icon in the Tree View changes to either Wire (N-Linear), Curve or Polygon.

How to recreate:

- Launch FreeCAD and in the Start page choose 2D Draft. Make sure the Tree View is visible.
- Create a Line (LI shortcut) and see a new Part icon being added in the Tree View under your Document.
- Cancel the action (Esc), the Part icon in the Tree View disappears.
- Create a Circle (CI shortcut) and see that no Part icon is added in the Tree View, until you finish your Task (e.g. set the center and the radius), then a Curve icon is added in the Tree View.

Proposed solution:

- Make all Draft Tools consistent and do not add a temporary Part_3D_object icon in the Tree View until the task is finished. Many Draft Tools already perform that way (see above the list). Arch WB and Sketcher WB perform that way.
- Or: make all Draft Tools consistent and add the right icon (Wire/N-Linear for Line, Polyline, etc.) as soon as the Tool is called. Part Design WB and TechDraw WB perform that way.

### Full version info

```shell
OS: GNU/Linux Ubuntu 22.04 LTS based distro (DE: GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36997 (Git)
Build type: Release
Branch: main
Hash: bff1efd665dc2f553ac57e5bef35c44a879999c2
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.5.1
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

farleyrunkel, 2024-04-29:
Hello, @marcuspollio , could you please add more images.

marcuspollio, 2024-04-29:
Hi @farleyrunkel just added some screenshots in the collapsed section of the first post.

yorikvanhavre, 2024-04-30:
That temporary object was there basically so the tool could snap to itself. Which I think is still a good thing. But indeed the temp object is annoying. Two possibilities:

1. We remove the temp object and adapt the snap system so it can work with a non-existing object (the polyline being drawn) - long path
2. We make the temp object invisible - short path

I'm inclined to the short path. What do you think @Roy-043 ?

Roy-043, 2024-04-30:
I agree.

closed 2024-05-02 by Roy-043

marcuspollio, 2024-05-02:
@yorikvanhavre and @Roy-043

Thank you for having a look !

With PR 13778 (or commit b45bf9b) applied, it only fixes this issue for the Line and the Cubic Bézier Curve tools.

For the Polyline, the B-Spline and the Bézier Curve, now there is an even more odd item in the tree when using the Tool :

[image]

How to reproduce:

- Launch FreeCAD and in the Start page choose 2D Draft. Make sure the Tree View is visible.
- Create a Polyline, a B-Spline or a Bézier Curve, click the base point in the 3D view (n°0 on the screenshot above), nothing happens, set a second point (n°1), this odd icon appears in the Tree View, then continue (n°2,3 and so on) until your Object is fully set. When double-clicking (or closing in the Task panel), the icon in the Tree View changes into a Wire or BSpline.

Any clue why is it the case ?

PR #13778 (2024-05-02): Draft: Do not show temporary line object in tree
The Draft line/polyline tool creates a temp object while drawing, for self-snapping, that should not appear in the tree.

Fixes #13700
commit b731b8a9 (2024-05-02): Draft: Do not show temporary line object in tree - fixes #13700
commit 22dc1721 (2024-05-06): Draft: Do not show temporary line object in tree (additional)

#13778 introduced `self.obj.ViewObject.ShowInTree = False` to hide the temporary object. But objects that are hidden in the tree are displayed in the tree when they are selected in the 3D view. This selection occurs during commands that take more than 2 points. The solution is to change `self.obj.ViewObject.Selectable` on mouse button up/down.

Fixes #13700.