FreeCAD/FreeCAD #13712: Measurement: Support of individual picking points for distance

maxwxyz, 2024-04-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently when selecting two lines, surfaces or features in the new unified measurement tool, the points to calculate a distance are selected automatically by the tool, it seems to choose the closest ones.
It should be possible to select individual picking points that are on the selected features, e.g. select the point on a line or surface from which the distance should be calculated.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hlorus, 2024-05-01:
Isn't this the same as #13708?

maxwxyz, 2024-05-01:
@hlorus no, this issue here is meant to pick a random, user chosen point, like the current Std_MeasureDistance tool can do, which is the last to replace.

hlorus, 2024-05-01:
Thanks for clarifying!

PR #13810 (2024-06-17): Integrate Std_MeasureDistance into unified measurement facility
Currently the Std_MeasureDistance is still present as it's functionality wasn't completly covered by UMF's MeasureDistance. This was because the UMF version would only measure between actual geometry elements rather than  the locations where the user is clicking.

This PR adds a new measure type "MeasureDistanceDetached" to UMF which allows to measure a distance between two locations on the surface of objects. 
Also the old command "Std_MeasureDistance" along with the feature "App/MeasureDistance.h" and it's viewprovider "Gui/Meas [... cut]
closed 2024-06-17 by chennes