FreeCAD/FreeCAD #13714: Installed preference packs no longer shown in preferences

furgo16, 2024-04-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When going to <kbd>Edit</kbd> > <kbd>Preferences</kbd> > <kbd>General</kbd>, the list of installed preference packs is empty. Neither the built-in nor the manually installed packs are shown. The <kbd>Manage...</kbd> button does work, in that the dialog that appears upon clicking on it shows all themes (built-in and manual).

This side by side comparison with a screenshot seems to hint to a regression. FreeCAD 0.21 did show the preference packs. On my system I can reproduce this behavior with both the snap and Debian (freecad-daily PPA) packages.

|FreeCAD 0.21| FreeCAD 0.22 |
| --- | --- |
|[image] | [image]|

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36999 (Git) Snap 1001
Build type: Release
Branch: main
Hash: ed77603af999525b65dcfc4a41337f15a052c9ef
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.4.20
  * BIM 2021.12.0
  * Curves 0.6.35
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-29:
I cannot reproduce this in Windows 11, my preference pack from 0.21.2 is also visible in 0.22

chennes, 2024-04-30:
Themes now show up in the Theme drop-down, right? I thought that was why we added the Theme addon type in the first place.

Roy-043, 2024-04-30:
@furgo16 This may be pathing issue. Do you experience the same with installed external workbenches?

furgo16, 2024-05-01:
> Themes now show up in the Theme drop-down, right? I thought that was why we added the Theme addon type in the first place.

@chennes 

In 0.22, themes show up:
1. :white_check_mark: under <kbd>Preferences</kbd> > <kbd>General</kbd>, **Application** section, **Theme:** dropdown. 
1. :x: under <kbd>Preferences</kbd> > <kbd>General</kbd>, **Preference packs** section (although not for me here, hence this bug)
1. :white_check_mark: under <kbd>Preferences </kbd>> <kbd>General</kbd>, **Preference packs** section, <kbd>Manage...</kbd> button
 
In 0.21 themes showed up:
1. :white_check_mark: under <kbd>Preferences</kbd> > <kbd>General</kbd>, **Application** section, **Stylesheet:** dropdown.
1. :white_check_mark: under <kbd>Preferences</kbd> > <kbd>General</kbd>, **Preference packs** section
1. :white_check_mark: under <kbd>Preferences </kbd>> <kbd>General</kbd>, **Preference packs** section, <kbd>Manage...</kbd> button

> This may be pathing issue. Do you experience the same with installed external workbenches?

@Roy-043 I'm not sure I quite follow. The external workbenches I have installed show up correctly under <kbd>Preferences</kbd> > <kbd>Workbenches</kbd> > <kbd>Available Workbenches</kbd>. Let me know if that answers the question.

I initially thought that this might be a packaging issue, but I've observed that I get this same behavior on the snap package, the Debian package, and in a local build from a fresh checkout.

chennes, 2024-05-01:
> In 0.22, themes show up:
>     1. ✅ under Preferences > General, **Application** section, **Theme:** dropdown.
>     2. ❌ under Preferences > General, **Preference packs** section (although not for me here, hence this bug)
>     3. ✅ under Preferences > General, **Preference packs** section, Manage... button

This was a specific (and deliberate) change to the way Theme preference packs are handled. 0.21 had no conception of a _theme_, and only showed _stylesheets_ in its list. In 0.22 dev, we now have the ability for an Addon to declare itself a "theme" (generally a stylesheet combined with a set of user preferences). We've eliminated user-facing bare Qt stylesheets entirely, and internally all themes are preference packs. So when you install an addon declared as a Theme, it will show up only in the themes menu (though its installation can still be managed by the preference packs manager, since technically it's a combination of a preference pack and a stylesheet). If this behavior is not to your liking, a discussion with the @FreeCAD/design-working-group may be fruitful to explore other options.

furgo16, 2024-05-01:
Thanks for the clarification. I understand the reasoning, but from https://github.com/FreeCAD/FreeCAD/issues/13714#issuecomment-2083452151 it would seem that on Windows the preference pack themes show up both in the Theme dropdown and in the Preference packs list.

Also related: https://github.com/FreeCAD/FreeCAD/issues/13748

MisterMakerNL, 2024-07-08:
I don't see themes in the preference list on windows.
It works as should.
There are tho theme's that haven't added the theme tag and so show up.

furgo16, 2024-07-09:
I'm closing this issue, as it seems it's a design decision and it works as intended.

I still think it's a bit confusing for users to not show the installed themes in the Preference Packs section. Especially as they are named Preference Packs on installation ([no mention of themes in the Add-on Manager](https://github.com/FreeCAD/FreeCAD/issues/15281)) and then subsequently referred to as Themes once installed.

It took me a while to figure it all out, but now that I understand it, I admit do not feel too strongly about it. There are more pressing issues, so let's agree to disagree and move on :-)

closed 2024-07-09 by furgo16