FreeCAD/FreeCAD #13715: Measurement: Measure mass with material of the object

maxwxyz, 2024-04-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The unified measurement tool supports the measurement of volumes for the selected objects / bodies.
With the material of the object the measurement tool should also support the measurement of mass.
Using the density of the selected object from the object's material, it should calculate the mass by multiplying it with the volume.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-04-29:
Inertia should be part of this as well. In other software it is displayed alongside the mass, which is good because they are closely tied together.

The most basic implementation would be computing the inertia matrix around the 3 principal axis. A better feature would be to allow to select a point as the origin (with keeping the main directions parallel so just a translation), a LCS, or even just a line (so the inertia around that).

maxwxyz, 2024-04-29:
Please create a new issue for this, you can link it here.

pierreporte, 2024-04-29:
@maxwxyz done with #13719. I mentioned the center of gravity. Not sure if it belongs here or there.

maxwxyz, 2024-04-30:
COM is already supported

Panda631, 2025-08-05:
I would kindly bring up this topic.
FC should definitifely support this info.