FreeCAD/FreeCAD #13716: Measurement: Support thickness measurement

maxwxyz, 2024-04-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Mentioned here: https://github.com/FreeCAD/FreeCAD/pull/9750#issuecomment-2002003690

The new unified measurement tool should support the measurement type thickness to measure the thickness on any given point on an object. The thickness could be calculated using the normal vector dierection of the picking point.
[image]

Example from CATIA:
[image]

@pierreporte FYI

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Anything else?

Related:
- https://github.com/FreeCAD/FreeCAD/issues/13712

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tobiasfalk, 2024-04-30:
It would also be nice if the maximum and minimum Thiknes would be displayed, maybe with a angle to the normal vector

pierreporte, 2024-04-30:
@tobiasfalk How would you define minimum and maximum thickness?

tobiasfalk, 2024-05-01:
@pierreporte I mean it in two ways, the first st that a non uniform body may has a spot where the thickness is at a minimum and a maximum.(this could be done if a surface is selected)
The second is that the thickness may change depending on the angle and that once again in non uniform body's the Normal Vector could be the maximum(same as the cross section of a cube) or something complete else, there it could be interesting in what ange to the NV the maximum and minimum are.
I understand that this will make it a lot more complicated.

pierreporte, 2024-05-01:
I still don’t get what the use case would be.

Thickness analysis would be great though, with the part colored like in FEM to indicate thickness below every point of the surface. This is out of scope of this issue.

tobiasfalk, 2024-05-02:
[image]

maxwxyz, 2024-05-02:
Current issue is meant to get the thickness under the picking point of the surface. So calculate the normal direction at the picking point and follow that direction until another surface of that body is hit.
Min/max/mean could also be a use case (your 1.) but this is fairly complex on curved surfaces. For your point 2. i currently do not see a use case but this could still be measured by picking the individual start/end point as workaround.

pierreporte, 2024-05-05:
I think I got it. There are two kind of thicknesses:

1. normal thickness, that is measured by the length of the segment normal to the surface at the selected point;
2. total thickness, that is measured by the length of the smallest segment starting at the selected point and not necessarily normal to the surface.

The first one is the original idea illustrated by @maxwxyz’s pictures. The second one is usually seen with thickness analysis where the part is fully colored with specific shades corresponding to known thicknesses, and where the edges or corners have usually zero thickness. The total thickness is useful for manufacturing because some materials or processes need a minimum thickness (like 3D printing), or constant thickness (like injection molding).

Thus I think that the thickness measurement tool should be split in two: normal and total. As well, a potential thickness analysis feature could  allow the user to compute both normal and total thicknesses.

leoheck, 2025-11-03:
Hi guys, any progress on this one?

pierreporte, 2025-11-03:
@leoheck not on this specific feature. @ryankembrey is working on the DFM workbench that will have thickness measurement, but the standalone feature will not be implemented before the workbench works. If you want to tackle this, go ahead.

pierreporte, 2026-08-18:
@ryankembrey would you be interested in this, considering you resolved https://github.com/ryankembrey/FreeCAD-DFM-Workbench/issues/81?

ryankembrey, 2026-08-19:
I might give it a go on the weekend if I find some time, but I've never looked at measurement module before so not sure what to expect in terms of implementation.

The calculation of the thickness values should be relatively simple since I'll just port my python OCCT code to Cpp.

 Do we want both thickness methods? (Ray and sphere) @pierreporte

pierreporte, 2026-08-19:
Just ray, as illustrated. The idea is to have the thickness at a given point in the normal direction.

ryankembrey, 2026-08-31:
I did end up having a go at getting it working on that weekend. I got it 99% working but I had a few bugs I wasn't able to fix so put it aside for now. I will come back to it another time in nearish future.

pierreporte, 2026-08-31:
@ryankembrey could you open a draft PR? We could test it and find more issue for when you’re ready again to work on it.