FreeCAD/FreeCAD #13717: Annotation Label position has no attachment support

maxwxyz, 2024-04-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The `AnnotationLabel` when created has no attachment support for the origin of the annotation line(s).
It is not possible to set the position and keep it updated when the annotation is placed at a selected geometry or object.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36729 (Git)
Build type: Release
Branch: main
Hash: 6ca35709ddc7f95fb84eef4c24973dc489e5acde
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
  * CfdOF 1.25.11
  * Curves 0.6.35
  * dodo 1.0.1
  * fasteners 0.5.20
  * fcVM-WB
  * freecad.gears 1.2.0
  * OpenTheme 2024.4.20
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-18:
@maxwxyz Is this a Measurement related ticket ?

maxwxyz, 2024-08-18:
@luzpaz no