FreeCAD/FreeCAD #13718: Measurement: Annotation is fixed and not moved with the referenced object

maxwxyz, 2024-04-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When a measurement (https://github.com/FreeCAD/FreeCAD/pull/9750) is saved as annotation the location of the displayed position in the 3D view is fixed and won't update if the referenced object / geometry is moved.
[image]

The expected behavior would be, that the measurement needs an update/recompute and will be displayed at the new location of the reference points.

Maybe this is an issue, as the annotation label has no attachment support:
- https://github.com/FreeCAD/FreeCAD/issues/13717

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Anything else?

- https://github.com/FreeCAD/FreeCAD/issues/13537
- https://github.com/FreeCAD/FreeCAD/issues/13717

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hlorus, 2024-05-03:
This seems to be a missing recompute. For me it updates if i trigger it manually. That being said it seems to work in simple cases e.g. a measurement on a part Box.

hlorus, 2024-05-03:
The problem seems to be that we only get signals of our subject but not from it's container (e.g. PD Body). @WandererFan maybe you have an idea how we could solve this?

maxwxyz, 2024-07-15:
@wwmayer could you support here?

wwmayer, 2024-07-25:
The measurement object has some kind of caching problem.
1. Create a part box
2. Add a distance measurement
3. Select Vertex1 and Vertex5 of the cube
4. Save the distance measurement
5. Set the length of the cube to 20 mm and recompute the document
6. The label shows the correct value but its length is 10 mm
7. Change the length of the cube back to 10 mm
8. The label shows the correct value but its length is now 20 mm

maxwxyz, 2024-08-12:
any news on this issue?

hlorus, 2024-08-13:
Sorry for the silence. AFAIK this is an issue with the distance measurement only, should be solved with #15874.

PR #15874 (2024-08-19): [MeasureGui] Fix measure distance viewprovider update
This addresses the issue described in #13718 where the drawing of a distance measurement doesn't correctly update when the underlying geometry moves or is changed.

Fixes #13718
closed 2024-08-19 by WandererFan