FreeCAD/FreeCAD #13719: Measurement: Measure inertia with material and coordinate system

pierreporte, 2024-04-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With the material of the object the measurement tool should also support the measurement of inertia. The whole matrix should be displayed, plus the principal moments of inertia and its associated coordinate system, possibly different than the base one, with the origin at the center of gravity.

Several implementations are possible and all are desirable:

- [ ] Most basic: use only the base coordinate system.
- [ ] Use a point as the origin. The coordinate system is then a translation of the base one.
- [ ] Use a local coordinate system.
- [ ] Use a single line to compute the inertia around it. 

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Anything else?

Needs mass calculation, see #13715.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-04-30:
This measurement should also pick volumes by default, see issue #13732.

maxwxyz, 2024-08-28:
This is implemented in the external WB FCInfo and also COM