FreeCAD/FreeCAD #13731: Measurement: Center of mass annotation should be a datum point

pierreporte, 2024-04-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, the center of mass measurement displays a text containing the coordinates of the center of mass with a line pointing to it. The actual point can’t be picked nor used directly unless created manually using the given values, that have a limited precision. The current annotation is fine for a simple measure, but when the user wants it to be persistent, it should be a datum point. Its position can be read with the property panel.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37063 (Git) AppImage
Build type: Release
Branch: main
Hash: 680792030fd664c94829e2c1b78b21788e8d9879
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * OpenTheme 2024.4.20
  * Curves 0.6.35
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-30:
Maybe you can rephrase. I know this behavior from Catia that you can create geometry from measurements. Also axes for example.

pierreporte, 2024-04-30:
Yes this is the idea.

maxwxyz, 2024-10-03:
this could be made with the implementation of https://github.com/FreeCAD/FreeCAD/pull/16675