FreeCAD/FreeCAD #13732: Measurement: Center of mass should pick volumes by default

pierreporte, 2024-04-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The center of mass of a solid can only be calculated when it is selected in the tree. When clicking in the 3D view, only surfaces and lines can have their center of mass calculated. While it can be useful, it is rarely the intent.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37063 (Git) AppImage
Build type: Release
Branch: main
Hash: 680792030fd664c94829e2c1b78b21788e8d9879
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * OpenTheme 2024.4.20
  * Curves 0.6.35
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-30:
I think currently you can select one or more features (faces,  lines) and get info about their center or midpoint. This current behavior would be lost.

pierreporte, 2024-04-30:
It won’t be lost if it allows to select everything else. It would just be a selection filter similar to `Part_SelectFilter` enabled by default. It could in fact be part of `Part_SelectFilter`, expanded accordingly (we could even imagine allowing to filter parts and assemblies).

luzpaz, 2024-07-12:
Please open a forum discussion and link to it from this ticket, TIA

pierreporte, 2024-08-11:
I just created a forum thread: https://forum.freecad.org/viewtopic.php?t=89717