FreeCAD/FreeCAD #13733: Measurement: task panel doesn’t allow to edit selection

pierreporte, 2024-04-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The current measurement task panel only shows mode and result:

[image]

It would be more practical to have a similar UI to the attachment panel so that the user could edit the selection. It would help correct mistakes in case of misclicks or when the desired result is not achieved.

[image]

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37063 (Git) AppImage
Build type: Release
Branch: main
Hash: 680792030fd664c94829e2c1b78b21788e8d9879
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * OpenTheme 2024.4.20
  * Curves 0.6.35
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-04-30:
The “tape measurement” features had it:

[image]

maxwxyz, 2024-04-30:
I think this needs more discussion. The current selection is easy to use. Select and click again to deselect. Click in the empty space and it is cleared or use the reset button

pierreporte, 2024-05-01:
An alternative way could be to use measurement annotations to make selection more obvious in 3D, with some properties attached (like point/plane/line/circle or what part is belongs to). Clicking on those would unselect the geometry.